EddyVerbruggen/nativescript-nfc #2: Plugin affects navigation in Nativescript with Angular app after installation

neilsb, 2017-04-03:
After installing the nativescript-nfc angular plugin navigation stopped working after the back button was pressed on an android device.

Both **[nsRouterLink]** in the html or **routerExtensions.navigate()** in typescript fail to produce navigation or an error message.

Can be reproduced simply by creating a new app from the template and installing teh nativescript-nfc plugin.

    tns create testApp --ng
    tns plugin add nativescript-nfc

After the plugin installation, the first tap to view item detail will work fine, but after tapping back further taps will not result in any item detail being displayed.

EddyVerbruggen, 2017-04-04:
Hi,

I created the app according to your instructions and added the plugin. Then ran `tns run ios` and navigated back and forth.. doesn't break. Do I need to do anything else to reproduce it? Like importing the plugin for instance?

neilsb, 2017-04-05:
I've not tested on ios yet, but should be able to do that this evening and let you know, but I can reproduce it on android.

Simply adding the plugin is enough to cause the error, and as I result I suspect that the problem is to do with hooking into the Android Activities and Intents, since that part exectuted on application startup as soon as the plugin has been added to the project.

EddyVerbruggen, 2017-04-05:
I can imagine that being true, but I just tried the same on Android 7 and there's no problem there either. So maybe I'm missing a step.

neilsb, 2017-04-05:
Hi,

Myself and my colleague both experienced the same problem on both the android emulator and android devices (versions 4, 6 and 7).  Let me try ios later on and I'll spin up a new VM and re-check on Android with a clean install and I'll get back to you.

neilsb, 2017-04-07:
I can confirm the issue does not show itself when running in the iPhone simulator.

On a Windows 10, with a fresh install following the Nativescript documentation the issue can be reproduced when running on the android emulator or a connected android device however.

EddyVerbruggen, 2017-04-07:
Let me run it on Android this weekend and get back to you. Thanks for your tests!

closed 2017-04-10 by EddyVerbruggen