EddyVerbruggen/nativescript-nfc #4: Add iOS support

EddyVerbruggen, 2017-06-05:
Available on iOS 11, on an iPhone 7 (Plus): https://twitter.com/eddyverbruggen/status/871844669956784129

sublimator, 2017-06-22:
How did you go with this ?

EddyVerbruggen, 2017-06-22:
Waiting for the next iOS 11 beta because I need to use the wife's iPhone 7 for this and I don't want to brick her primary phone with the seemingly unstable first beta.

sublimator, 2017-06-22:
Fair enough :)

I just tried a hello world NS app but had some issues with livesync etc. It
did a full rebuild on save and then complained about an app with the same
name already installed.

sublimator, 2017-06-23:
Beta 2 is out by the way :)  I just saw
http://adcdownload.apple.com/Developer_Tools/Xcode_9_beta_2/Release_Notes_for_Xcode_9_beta_2.pdf

EddyVerbruggen, 2017-08-03:
I've got the basics working now. The entitlements handling seems to have changed a bit in Xcode 9, or it's due to the fact that it's in beta (4). The main task left is decoding the returned data from the NDEF tag.. hopefully I'll be able to do that soon. All code is on GitHub btw.

sublimator, 2017-08-04:
There's this: https://github.com/don/ndef-js

sublimator, 2017-08-04:
We're stuck in our loveless marriage with ionic for the moment, but still
looking longingly now and then at NS haha :)

closed 2017-12-08 by EddyVerbruggen