EddyVerbruggen/nativescript-nfc #6: Demo not working on Android

daaaaa, 2017-08-21:
This plugin seemed promising to me, but unfortunately I am not able to make it work.

I have a Samsung Galaxy A3.
I am using the following type of nfc tag: [Link](https://www.shopnfc.it/it/braccialetti-nfc/54-195-bracciali-nfc-in-silicone-premium.html#/19-chip_nfc-ntag213/29-personalizzazione-no)
I have use the command `npm run test.android`. The app builds correctly and appear loads on the device.
Button nfc avaiable and enables work (show true when activated, false otherwise).
But neither set ndef listener, nor set tag listener seem to detect my nfc tags.

Moreover I also noticed that the image having the text 'ready to scan' on yout github and on your video with the iphone, does not appear.

All others application on the market detect those nfc tags, but your demo was not able.

Is there anything I am doing wrong?

EddyVerbruggen, 2017-08-21:
You should use `npm run demo.android`, the `test` script is for running tests on CI.

The "ready to scan" UI is the iOS UI for scanning tags, there's no such thing (needed) for Android.

I don't have such a tag so I can't test your case..

daaaaa, 2017-08-21:
Yes, sorry, I meant `npm run demo.android`.

That tag is a standard one. It does not detect anything unfortunately. 

Otherwise could you show me a compatible one? 
Tomorrow I will try with another phone and I will let you know.

daaaaa, 2017-08-23:
Yesterday I tried the same operations using an LG Nexus 4. 

I also tried to use a complete different nfc device. This is still not detected by the demo app, but detected by all the apps on the Android store. 

I tried to implement by myself the api, but the result remains the same: `available` and `enable` apis work correctly, but the callbacks set by the listeners are never called. 

Is there a chance you can test the app on an Android device?

EddyVerbruggen, 2017-08-23:
Hi, that's really unfortunate to hear. I only have a Nexus 6P and 1 specific type of NDEF tag to test with. And that works fine.

Is there anything `adb logcat` can tell you while scanning the tag (with the ndef listener enabled)?

And please also try version 1.0.0 of this plugin to rule out the chance that I screwed something up in the last release..

daaaaa, 2017-08-24:
Today I had the opportunity to try quickly to build the demo app at version 1.0.0, but an error prevented me to build the app. 

In the next days I'll try to use another computer and I'll let you know. 

It's unfortunate I'm not able to use this plug in, the api's are so simple and well done that is exactly what I'm searching.

ikhsan017, 2017-09-19:
I have the same issue,

just run android demo, "nfc available" and "nfc enabled" both alert "true", but "set ndef listener" and "set tag listener" do nothing, "write *" button only alert "connection failed"

also, on my code

```
    this.nfc.setOnNdefDiscoveredListener((data: NfcNdefData) => {
            // breakpoint1
            if (data.message) {
                for (let m in data.message) {
                    let record = data.message[m];
                    console.log("Ndef discovered! Message record: " + record.payloadAsString);
                }
            }
        }).then(() => {
            // breakpoint2
            console.log("OnNdefDiscovered listener added");
        });
``` 

I put a breakpoint on breakpoint1 and breakpoint2, breakpoint2 is executed, but breakpoint1 never get executed when NFC tag is approached.

daaaaa, 2017-09-29:
Other friends of mine tried this plugin on android devices and V2 definitely does not work.
They was not able to try V1 like me because something is broken and the build does not start.

EddyVerbruggen, 2017-12-08:
Please try 2.1.0, I just published it on npm - or clone this repo and run the demo app (all its NativeScript dependencies are bumped to the latest versions).. it works great for me on iOS and Android.

closed 2017-12-08 by EddyVerbruggen

ukch, 2017-12-15:
@EddyVerbruggen The demo still does not work for me using the latest version. Here is what I did:

* Clone the latest `nativescript-nfc` from Github
* `cd nativescript-nfc/src && npm run demo.android`
* In app, check the 'NFC Available' and 'NFC Enabled': these both return true
* Press the 'Set NDEF Listener' button
* Touch the NFC tag to the phone's NFC reader: nothing happens

I have tested the same NFC tag on the same phone, using the [NFC Reader](https://play.google.com/store/apps/details?id=se.anyro.nfc_reader) app. Here are the results:
```
ID (hex): [HIDDEN]
ID (reversed hex): [HIDDEN]
ID (dec): [HIDDEN]
ID (reversed dec): [HIDDEN]
Technologies: NfcA, MifareUltralight, NdefFormatable
Mifare Ultralight type: Ultralight
```
Using Node v8.9.3 (v9.2.1 also tested), and Nativescript CLI 3.3.1.

EddyVerbruggen, 2017-12-15:
What Android device / version is that?

My Samsung S8 and Nexus 6P have no issues detecting a tag, but it could be that tag is not supported by ndef, maybe?

ukch, 2017-12-18:
It's a Sony Xperia FC111 running Android 7.0.

jdvos, 2017-12-27:
Library 2.1.0 is not working for me either.
Same effect as ukch : both nfc available and enabled = true, but listener does not handle the card.
With library 2.0.0 no problem.
Samsung Galaxy Tab Active2 with Android 7.1.1

EddyVerbruggen, 2017-12-27:
@jdvos If 2.0.0 did work, can you verify that you've made [this change](https://github.com/EddyVerbruggen/nativescript-nfc/tree/f9f2eadf7a1f174de46d0540fe4263bf1735c3c3#android-setup) with 2.1.0?

jdvos, 2017-12-27:
Missed the activity change.
Now version 2.1.0 works flawlessly.
Thank you.

cwjacobs, 2019-03-01:
Same result as jdvos when I made the change to the AndroidManifest. Thumbs up!!