EddyVerbruggen/nativescript-nfc #8: Webpack Issues Angular

skhye05, 2017-11-23:
### When i am installing the plugin and running webpack command i am getting the following error
`Execution failed for task ':asbg:generateBindings'.
> Process 'command 'C:\Program Files\Java\jdk1.8.0_144\bin\java.exe'' finished with non-zero exit value 1
`
### What could be the issue?? and how can I make it work??

EddyVerbruggen, 2017-11-23:
That's an incredibly unuseful error message.. can't help with that.

skhye05, 2017-11-23:
Did you ever run into an error with webpack while using the plugin with nativescript + angular??

EddyVerbruggen, 2017-11-23:
I have never done that with this plugin.

skhye05, 2017-11-23:
@EddyVerbruggen How can i exclude it from webpack?? and if i do will it work??

manojdcoder, 2017-11-28:
@EddyVerbruggen I'm using this plugin with Angular and WebPack, iOS complies just fine but there is known issue with [Android](https://github.com/NativeScript/nativescript-cli/issues/2901)

As @Pip3r4o suggested, I simply renamed the JavaProxy and updated same in manifest, everything was just perfect then. It may not be a feasible solution but at least the build is happening for now.

EddyVerbruggen, 2017-11-28:
@manojdcoder Would you be willing to do a PR to change that?

closed 2017-11-28 by EddyVerbruggen

EddyVerbruggen, 2017-11-28:
If it were trivial I'd do it immediately but I'm thinking about the implications still.

What if 2 plugins would need a changed activity name? So I don't really like this fix (although it was recommended in the issue you linked to)... so need some time to digest it.

manojdcoder, 2017-11-28:
@EddyVerbruggen Exactly that is the same reason I felt this is not a feasible solution. But for time being I had to go with this.

commit 3bf55d35 (2017-11-28): Merge pull request #9 from manojdcoder/master

Fixed #8 - Webpack issues
commit 6172970f (2017-11-28): Fixed #8 - Webpack issues