EddyVerbruggen/nativescript-nfc #10: card emulation

Nurgunkalol, 2017-12-01:
is any tutorial for card emulation of device ?
thx for advance

EddyVerbruggen, 2017-12-01:
Dunno, but if you find one, feel free to post it here.

closed 2017-12-01 by EddyVerbruggen

Nurgunkalol, 2017-12-05:
I am trying to do like this:
declare var android: any;
@JavaProxy("com.tns.MyHostAdpuService")
class MyHostAdpuService extends android.nfc.cardemulation.HostAdpuService { }

also added it to AndroidManifest but tns-nativescript-declarations doesnt see android.nfc.cardemulation
+ on testing with scanner its always shows "unable to instantiate service classnotfoundexception"

can anyone give me advance how to do it

Nurgunkalol, 2017-12-28:
Im done NFC HostApduService:

declare var android: any;

@javaproxy("com.tns.MyHostAdpuService")
class MyHostAdpuService extends android.nfc.cardemulation.HostAdpuService { }

onDeactivated(reason: number) { }

processCommandApdu(commandApdu: Int8Array, extras?: android.os.Bundle): { // your code}

works great!

xCyrusx, 2019-02-27:
Hi @Nurgunkalol I'm in the process of developing a NativeScript app that needs host-based card emulation (HCE) to pay in readers connected to arcade games and it seems like there is not much information about this topic. I noticed that you managed to create a functional app.
Would it be too much if I ask for a running code example if you have it at hand?

harfyt, 2023-05-08:
> android.nfc.cardemulation.HostApduService

Hello @n4lexeev, could you share a complete example of how did you manage to send an NDEF message ?