EddyVerbruggen/nativescript-nfc #13: Nativescript 3.4 Ng 5.* compatibility

ikhsan017, 2017-12-29:
Hi Eddy,

sorry to disturb in this holiday season, but it seems after I update my tns to 3.4, nativescript nfc is now broken.

So I tried to create new project using ```tns create sample --ng```, modify the sample page to looks like this, and got error.

It seems that we can not access application.android.foregroundActivity in NS 3.4 

template
```
<StackLayout class="page">
    <Button text="scan nfc" (tap)="addNfcListener()"></Button>
</StackLayout>
```

component
```
export class ItemsComponent implements OnInit {
    items: Item[];
    nfc: Nfc;

    // This pattern makes use of Angular’s dependency injection implementation to inject an instance of the ItemService service into this class.
    // Angular knows about this service because it is included in your app’s main NgModule, defined in app.module.ts.
    constructor(private itemService: ItemService) {
        this.nfc = new Nfc;
    }

    ngOnInit(): void {
        this.items = this.itemService.getItems();
    }

    addNfcListener() {
        this.nfc.setOnNdefDiscoveredListener((data: NfcNdefData) => {
            if (data.message) {
                const records = [];

                data.message.forEach((record) => {
                    records.push(record.payloadAsString);
                });

                this.nfc.setOnNdefDiscoveredListener(null).then(() => {
                    dialogs.alert(records[0]);
                });
            }
        }, {stopAfterFirstRead: true}).then(() => {
            dialogs.alert('listener stopped');
        }, (err) => {
            dialogs.alert(err);
        });
    }
}
```

error
```
JS: ERROR Error: Uncaught (in promise): TypeError: Cannot read property 'getClass' of undefined
JS: TypeError: Cannot read property 'getClass' of undefined
JS:     at new Nfc (file:///data/data/org.nativescript.nsnfcdemo/files/app/tns_modules/nativescript-nfc/nfc.js:268:127)
JS:     at new ItemsComponent (file:///data/data/org.nativescript.nsnfcdemo/files/app/item/items.component.js:12:20)
JS:     at createClass (file:///data/data/org.nativescript.nsnfcdemo/files/app/tns_modules/@angular/core/bundles/core.umd.js:12204:20)
JS:     at createDirectiveInstance (file:///data/data/org.nativescript.nsnfcdemo/files/app/tns_modules/@angular/core/bundles/core.umd.js:12051:37)
JS:     at createViewNodes (file:///data/data/org.nativescript.nsnfcdemo/files/app/tns_modules/@angular/core/bundles/core.umd.js:13489:53)
JS:     at createRootView (file:///data/data/org.nativescript.nsnfcdemo/files/app/tns_modules/@angular/core/bundles/core.umd.js:13379:5)
JS:     at callWithDebugContext (file:///data/data/org.nativescript.nsnfcdemo/files/app/tns_modules/@angular/core/bundles/core.umd.js:14780:42)
JS:     at Object.debugCreateRootView [as createRootView] (file:///data/data/org.nativescript.nsnfcdemo/files/app/tns_modules/@angular/core/bundles/core.umd.js:14081:12)
JS:     at ComponentFactory_.create (file:///data/data/org.nativescript.nsnfcdemo/files/app/tns_modules/@angular/core/bundles/core.umd.js:11000:46)
JS:     at ComponentFactoryBoundToModule.create (file:///data/data/org.nativescript.nsnfcdemo/files/app/tns_modules/@angular/core/bundles/core.umd.js:3961:29)
JS:     at ViewContainerRef_.createComponent (file:///data/data/org.nativescript.nsnfcdemo/files/app/tns_modules/@angular/core/bundles/core.umd.js:11197:62)
JS:     at PageRouterOutlet.activateOnGoForward (file:///data/data/org.nativescript.nsnfcdemo/files/app/tns_modules/nativescript-angular/router/page-router-outlet.js:246:44)
JS:     at PageRouterOutlet.activateWith (file:///data/data/org.nativescript.nsnfcdemo/files/app/tns_modules/nativescript-angular/router/page-router-outlet.js:235:14)
JS:     at ActivateRoutes.activateRoutes (file:///data/data/org.nativescript.nsnfcdemo/files/app/tns_modules/@angular/router/bundles/router.umd.js:5652:40)
JS:     at file:///data/data/org.nativescript.nsnfcdemo/files/app/tns_modules/@angular/router/bundles/router.umd.js:5592:19

```

EddyVerbruggen, 2017-12-31:
Thanks for the report, I'll take a look!

EddyVerbruggen, 2017-12-31:
Looks like the `Nfc()` constructor is now (with NativeScript 3.4) able to run before the `foregroundActivity` is set.

I've changed the internal logic to check for the `foregroundActivity`, and if not set, listen for the app launch event (which is guaranteed to fire after `foregroundActivity` was set.

Available in 2.1.1.

ikhsan017, 2018-01-01:
Thanks for quick response Eddy,

Nfc instantiation give no error, but somehow I get following error when I start adding ndefListener (using demo included in this repo)

```
System.err: java.lang.RuntimeException: Unable to resume activity {org.nativescript.plugindemo.nfc/com.tns.NativeScriptNfcActivity}: com.tns.NativeScriptException: 
System.err: Calling js method onStart failed
System.err: Error: java.lang.IllegalStateException: Foreground dispatch can only be enabled when your activity is resumed
System.err:     android.nfc.NfcAdapter.enableForegroundDispatch(NfcAdapter.java:1227)
System.err:     com.tns.Runtime.callJSMethodNative(Native Method)
System.err:     com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1088)
System.err:     com.tns.Runtime.callJSMethodImpl(Runtime.java:970)
System.err:     com.tns.Runtime.callJSMethod(Runtime.java:957)
System.err:     com.tns.Runtime.callJSMethod(Runtime.java:941)
System.err:     com.tns.Runtime.callJSMethod(Runtime.java:933)
System.err:     com.tns.NativeScriptNfcActivity.onStart(NativeScriptNfcActivity.java:38)
System.err:     android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1245)
System.err:     android.app.Activity.performStart(Activity.java:6286)
System.err:     android.app.Activity.performRestart(Activity.java:6332)
System.err:     android.app.Activity.performResume(Activity.java:6337)
System.err:     android.app.ActivityThread.performResumeActivity(ActivityThread.java:3292)
System.err:     android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3338)
System.err:     android.app.ActivityThread$H.handleMessage(ActivityThread.java:1546)
System.err:     android.os.Handler.dispatchMessage(Handler.java:111)
System.err:     android.os.Looper.loop(Looper.java:207)
System.err:     android.app.ActivityThread.main(ActivityThread.java:5763)
System.err:     java.lang.reflect.Method.invoke(Native Method)
System.err:     com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:789)
System.err:     com.android.internal.os.ZygoteInit.main(ZygoteInit.java:679)
System.err: File: "file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/nativescript-nfc/nfc.js, line: 263, column: 31
System.err: StackTrace: 
System.err: 	Frame: function:'attachNfcAdapter', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/nativescript-nfc/nfc.js', line: 263, column: 32
System.err: 	Frame: function:'Nfc', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/nativescript-nfc/nfc.js', line: 290, column: 13
System.err: 	Frame: function:'HelloWorldModel', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/main-view-model.js', line: 11, column: 21
System.err: 	Frame: function:'pageLoaded', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/main-page.js', line: 6, column: 27
System.err: 	Frame: function:'Observable.notify', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/tns-core-modules/data/observable/observable.js', line: 103, column: 23
System.err: 	Frame: function:'Observable._emit', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/tns-core-modules/data/observable/observable.js', line: 120, column: 18
System.err: 	Frame: function:'ViewBase.onLoaded', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/tns-core-modules/ui/core/view-base/view-base.js', line: 202, column: 14
System.err: 	Frame: function:'View.onLoaded', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/tns-core-modules/ui/core/view/view.js', line: 62, column: 35
System.err: 	Frame: function:'Page.onLoaded', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/tns-core-modules/ui/page/page.js', line: 105, column: 35
System.err: 	Frame: function:'', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/tns-core-modules/ui/core/view-base/view-base.js', line: 206, column: 19
System.err: 	Frame: function:'FrameBase.eachChildView', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/tns-core-modules/ui/frame/frame-common.js', line: 437, column: 13
System.err: 	Frame: function:'ViewCommon.eachChild', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/tns-core-modules/ui/core/view/view-common.js', line: 699, column: 14
System.err: 	Frame: function:'ViewBase._loadEachChild', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/tns-core-modules/ui/core/view-base/view-base.js', line: 205, column: 14
System.err: 	Frame: function:'ViewBase.onLoaded', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/tns-core-modules/ui/core/view-base/view-base.js', line: 201, column: 14
System.err: 	Frame: function:'View.onLoaded', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/tns-core-modules/ui/core/view/view.js', line: 62, column: 35
System.err: 	Frame: function:'FrameBase.onLoaded', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/tns-core-modules/ui/frame/frame-common.js', line: 136, column: 35
System.err: 	Frame: function:'ActivityCallbacksImplementation.onStart', file:'file:///data/data/org.nativescript.plugindemo.nfc/files/app/tns_modules/tns-core-modules/ui/frame/frame.js', line: 640, column: 22
[... cut ...]
System.err: 	at android.app.Activity.performResume(Activity.java:6337)
System.err: 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3292)
System.err: 	... 8 more
System.err: Caused by: java.lang.IllegalStateException: Foreground dispatch can only be enabled when your activity is resumed
System.err: 	at android.nfc.NfcAdapter.enableForegroundDispatch(NfcAdapter.java:1227)
System.err: 	... 20 more
ActivityManager: Process org.nativescript.plugindemo.nfc (pid 11108) has died
```

EddyVerbruggen, 2018-01-01:
Thanks, can you give 2.1.2 a try?

closed 2018-01-01 by EddyVerbruggen

ikhsan017, 2018-01-03:
Great, it now works as expected

Thank you very much :)

kriefsacha, 2019-01-23:
Still same issue here : 

```
JS: ERROR Error: Uncaught (in promise): Error: java.lang.IllegalStateException: Foreground dispatch can only be enabled when your activity is resumed
JS:     android.nfc.NfcAdapter.enableForegroundDispatch(NfcAdapter.java:1440)
JS:     com.tns.Runtime.callJSMethodNative(Native Method)
JS:     com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1120)
JS:     com.tns.Runtime.callJSMethodImpl(Runtime.java:1000)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:987)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:967)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:959)
JS:     com.tns.NativeScriptNfcActivity.onCreate(NativeScriptNfcActivity.java:18)
JS:     android.app.Activity.performCreate(Activity.java:6608)
JS:     android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1134)
JS:     android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3086)
JS:     android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3243)
JS:     android.app.ActivityThread.access$1000(ActivityThread.java:218)
JS:     android.app.ActivityThread$H....
```

with a service like this : 

```
export class NfcService {
  nfc: Nfc;
  constructor() {    
    this.nfc = new Nfc();

  }
}
```

GijsPost, 2019-02-27:
Still stuck with the same issue as @kriefsacha, @EddyVerbruggen Any clues?

kriefsacha, 2019-02-27:
I create a new instance every time i need @GijsPost ..

glydetech, 2019-03-12:
I'm seeing exactly this same issue with latest version of all software components.

It appears to happen consistently on first-run after compiling and installing the app. I'm using nativescript-angular and webpack. I've tried instantiating in a public variable, constructor and ngOnInit, but the error still appears.

Error occurs on instantiation:
this.nfc = new Nfc();

Not sure where to go with this apart from perhaps catching the error and ignoring it. Any further help would be appreciated.

EddyVerbruggen, 2019-03-12:
Share a repo and I might find some time to look into it.

kriefsacha, 2019-03-12:
@glydetech as i saied i used a new instance that i created everytime in a function and it worked..