EddyVerbruggen/nativescript-nfc #16: On android "New Tag Collected" takes over the app when reading a card

mm-ns, 2018-01-05:
As per title, when i scan my tag (NDEF) on android normally the tag would be read by the OS "New Tag Collected" showing what i have written. My code, where the setOnNdefDiscoveredListener is, doesn't ever get a callback to print any data.. 

Can someone tell me why ?

EddyVerbruggen, 2018-01-05:
Did you do this? https://github.com/EddyVerbruggen/nativescript-nfc/tree/9430e1f175e6066d9a9510cb9f03cdacdca1895f#android-setup

mm-ns, 2018-01-05:
@EddyVerbruggen yes :) And it opens "New tag collected"

This is my manifest.xml

`
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
	package="__PACKAGE__"
	android:versionCode="1"
	android:versionName="1.0">

	<supports-screens
		android:smallScreens="true"
		android:normalScreens="true"
		android:largeScreens="true"
		android:xlargeScreens="true"/>

	<uses-sdk
		android:minSdkVersion="17"
		android:targetSdkVersion="__APILEVEL__"/>

	<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE"/>
	<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"/>
	<uses-permission android:name="android.permission.INTERNET"/>

	<application
		android:name="com.tns.NativeScriptApplication"
		android:allowBackup="true"
		android:icon="@drawable/icon"
		android:label="@string/app_name"
		android:theme="@style/AppTheme">
		<activity
			android:name="com.tns.NativeScriptNfcActivity"
			android:label="@string/title_activity_kimera"
			android:configChanges="keyboardHidden|orientation|screenSize">

			<intent-filter>
				<action android:name="android.intent.action.MAIN" />
				
				<category android:name="android.intent.category.LAUNCHER" />
			</intent-filter>
			
			
		</activity>
		<activity android:name="com.tns.ErrorReportActivity"/>
	</application>
</manifest>
`

mm-ns, 2018-01-05:
PS: I'm using angular with {N}

mm-ns, 2018-01-05:
seems like updating {N} to 3.4.0 did the trick..

closed 2018-01-05 by mm-ns