EddyVerbruggen/nativescript-nfc #24: getClass of undefined error (android)

spencercap, 2018-09-10:
hi, 
just trying to get this plugin up and running with my nativescript-vue project for android. following the installation instructions i reached this: `Error is: Cannot find module 'nativescript-custom-entitlements/lib/before-prepare.js’.` which i fixed by installing the [nativescript-custom-entitlements package](https://www.npmjs.com/package/nativescript-custom-entitlements) but carrying on, i cannot run the Nfc constructor `var nfc = new Nfc();` (after `var Nfc = require("nativescript-nfc").Nfc;` of course) without getting this getClass error:
```
System.err: java.lang.RuntimeException: Unable to create application com.tns.NativeScriptApplication: com.tns.NativeScriptException:
System.err:
System.err: Error calling module function
System.err:
System.err: Error calling module function
System.err:
System.err: TypeError: Cannot read property 'getClass' of undefined
System.err: File: "file:///data/data/org.***/files/app/vendor.js, line: 441, column: 67
System.err:
System.err: StackTrace:
System.err:     Frame: function:'Nfc', file:'file:///data/data/org.***/files/app/vendor.js', line: 441, column: 68
System.err:     Frame: function:'', file:'file:///data/data/org.***/files/app/bundle.js', line: 1346, column: 11
System.err:     Frame: function:'./main.js', file:'file:///data/data/org.***/files/app/bundle.js', line: 1368, column: 30
System.err:     Frame: function:'__webpack_require__', file:'file:///data/data/org.***/files/app/bundle.js', line: 76, column: 30
System.err:     Frame: function:'checkDeferredModules', file:'file:///data/data/org.***/files/app/bundle.js', line: 45, column: 23
System.err:     Frame: function:'module.exports.../node_modules/mini-css-extract-plugin/dist/loader.js!../node_modules/css-loader/index.js?!../node_modules/vue-loader/lib/loaders/stylePostLoader.js!../node_modules/vue-loader/lib/index.js?!./components/HelloWorld.vue?vue&type=style&index=0&id=763db97b&scoped=true&lang=css&', file:'file:///data/data/org.***/files/app/bundle.js', line: 149, column: 18
System.err:     Frame: function:'', file:'file:///data/data/org.***/files/app/bundle.js', line: 152, column: 10
System.err:     Frame: function:'require', file:'', line: 1, column: 266
System.err:     Frame: function:'', file:'file:///data/data/org.***/files/app/starter.js', line: 2, column: 1
System.err:     Frame: function:'require', file:'', line: 1, column: 266
```

EddyVerbruggen, 2018-09-10:
Hi, `nativescript-custom-entitlements` [is a dependency of this plugin](https://github.com/EddyVerbruggen/nativescript-nfc/blob/9430e1f175e6066d9a9510cb9f03cdacdca1895f/src/package.json#L61) so if that's not installed that's a red flag indicating `npm i` didn't run in the folder this plugin is installed. Please fix that before continuing. It's probably mentioned somewhere in the Vue template you're using.

closed 2018-09-10 by EddyVerbruggen

spencercap, 2018-09-10:
hi eddy, 
totally get the dependency issue and that's easily solved (sorry for bringing up two issues), but as the console reads, i'm having trouble even instantiating the plugin with `var nfc = new Nfc();` (for android). would adding this plugin to nativescript-vue project have anything to do with the error?

EddyVerbruggen, 2018-09-10:
Hi, no problem what you've reported, and indeed it must be related to how the Vue template/project is organized as the demo in this repo works just fine. Perhaps it's best to start there and see what's different? To run it, do:

```bash
git clone https://github.com/EddyVerbruggen/nativescript-nfc
cd nativescript-nfc/src
npm run setupandinstall .  # I think, not sure from the top of my head
npm run demo.android
```

spencercap, 2018-09-14:
thanks eddy - i was able to get the demo app up and running with ease.
perhaps i am not setting things up correctly in my project. can you elaborate on the android setup?
i try this step: `Update the activity entry in your App_Resources/Android/AndroidManifest.xml file` by replacing the default `com.tns.NativeScriptActivity` with `com.tns.NativeScriptNfcActivity` but i get the same getClass error or another not found class error. i can verify that the plugin is being required correctly but the constructor breaks things for me, any thoughts on where to start?

spencercap, 2018-09-14:
solution! 

after a bit of tinkering, i found that this error is actually an issue of using this plugin with the [vue-cli-template](https://github.com/nativescript-vue/vue-cli-template). i'm not exactly sure how/where the scope and/or reference to Nfc is lost, but i was not about the initialize Nfc with `var nfc = new Nfc();` in `main.js`. 
i did manage to get things working when requiring the plugin in the script tag of a vue single page component, then creating a scoped reference to `this.nfc` in the `created` hook like so:

```html
<script>
  var Nfc = require("nativescript-nfc").Nfc;

  export default {

    created: function() {

      this.nfc = new Nfc();

      this.nfc.available().then(
        function(avail) {
          console.log(avail ? "Yes Available" : "No Available");
        }
      );
      this.nfc.enabled().then(
        function(on) {
          console.log(on ? "Yes Enabled" : "No Enabled");
        }
      );

    },
    
    methods: {
      listen: function (event) {
        console.log(this.nfc); // is it there?

        this.nfc.setOnNdefDiscoveredListener(function(data) {
          if (data.message) {
            for (let m in data.message) {
              let record = data.message[m];
              console.log("Ndef discovered! Message record: " + record.payloadAsString);
            }
          }
        }).then(
          function() {
            console.log("OnNdefDiscovered listener added");
          }
        );
      }
    }
  };

</script>
```
i could then use the `this.nfc` object inside methods of the given component so i tied `listen` to a button click. 

many thanks to @youraerials

pdipasqua, 2018-12-15:
Hi @spencercap,
I have some issue with nfc and vue. I think that the issues are referenced to androidmanifest and webpack config because in the first i have some errors if add NativeScriptNfcActivity and for the second in node_modules doesn't exist the file nfc-activity.android.ts !!! My code is here #28 .
Thanks in advance. I hope you can help me! Bye

spencercap, 2018-12-15:
@pdipasqua did you try requiring Nfc like about? instead of importing Nfc from nativescript-nfc like your issue #28 ? 

I am only able to get NFC working within a single .vue component.

rebz, 2019-06-24:
I too am getting this error with Android. I am not using Typescript, but I am unable to `import` or `require` the package unless I do it within a script tag of some component; which is what I do not want to do.

This works perfectly fine for iOS, but building for Android just gives `Cannot read property 'getClass' of undefined`.

Is it because we are not using Typescript that this package does not work?