EddyVerbruggen/nativescript-nfc #28: Not fire on nativescript vue

pdipasqua, 2018-12-14:
Hi guys,
I have installed this plugin but the only methods that work are available and enabled.
This is my code on the mounted event:

```
            var nfc = new Nfc();

            nfc.available().then((avail) => {
                console.log(avail ? "Yes" : "No");
            });

            nfc.enabled().then((on) => {
                console.log(on ? "Yes" : "No");
            });
 

            nfc.setOnNdefDiscoveredListener(function(data) {
                if (data.message) {
                    for (let m in data.message) {
                    let record = data.message[m];
                    console.log("Ndef discovered! Message record: " + record.payloadAsString);
                    }
                }
                }).then(
                function() {
                    console.log("OnNdefDiscovered listener added");
                }
            );

            nfc.setOnTagDiscoveredListener(function(data) {
                console.log("Discovered a tag with ID " + data.id);
            }).then(function() {
                console.log("OnTagDiscovered listener added");
                }
            );

```
I declare NFC as:
import { Nfc } from "nativescript-nfc"

No errors and only logs on available and enabled on real android device but no read a mifare card.

Any suggest?
Thanks, P

hagai-shatz, 2018-12-19:
I have the same problem - NFC is enabled and listeners are set without any errors but no events are fired. The device makes the NFC read sound, but nothing in the app.

EddyVerbruggen, 2018-12-20:
Looks like a readme issue actually, but to kind of 'notify' users I've released a patch version (3.0.1) to npm. The change is this line (.js instead of .ts):

https://github.com/EddyVerbruggen/nativescript-nfc/blob/6dfa0ff4f77cab5ab7f494ac3055f728ce51d131/demo/webpack.config.js#L17

closed 2018-12-20 by EddyVerbruggen

EddyVerbruggen, 2018-12-20:
Also see this PR: https://github.com/pdipasqua/test-nfc/pull/1

Alberz, 2019-03-20:
I have the same problem. I added  `resolve(__dirname, "node_modules/nativescript-nfc/nfc-activity.android.js")` but events listener doens´t work.