EddyVerbruggen/nativescript-nfc #29: Listeners do not work on Angular w/ Android

GijsPost, 2019-02-28:
**Problem**:  
`setOnTagDiscoveredListener` and `setOnNdefDiscoveredListener` do not call their callbacks.
**Versions**:  
```
"nativescript": {
        "id": "org.nativescript.SurvivalRunApp",
        "tns-android": {
            "version": "5.2.1"
        },
        "tns-ios": {
            "version": "5.0.0"
        }
    },
    "dependencies": {
        "@angular/animations": "~7.0.0",
        "@angular/common": "~7.0.0",
        "@angular/compiler": "~7.0.0",
        "@angular/core": "~7.0.0",
        "@angular/forms": "~7.0.0",
        "@angular/http": "~7.0.0",
        "@angular/platform-browser": "~7.0.0",
        "@angular/platform-browser-dynamic": "~7.0.0",
        "@angular/router": "~7.0.0",
        "nativescript-angular": "~7.0.0",
        "nativescript-nfc": "^3.0.1",
        "nativescript-theme-core": "~1.0.4",
        "reflect-metadata": "~0.1.8",
        "rxjs": "~6.3.0",
        "tns-core-modules": "~5.0.2",
        "tns-platform-declarations": "^5.2.1",
        "zone.js": "^0.8.26"
    },
    "devDependencies": {
        "@nativescript/schematics": "~0.4.0",
        "nativescript-dev-typescript": "~0.7.0",
        "nativescript-dev-webpack": "~0.18.0",
        "@angular/compiler-cli": "~7.1.0",
        "@ngtools/webpack": "~7.1.0"
    },
```

**app.component.ts**:
```
export class AppComponent {

    nfc: Nfc;

    constructor(zone: NgZone) {
        if (android) {
            android.on(AndroidApplication.activityResumedEvent, (args: AndroidActivityEventData) => {
                console.log("Event: " + args.eventName + ", Activity: " + args.activity);
                this.nfc = new Nfc();

                this.nfc.available().then((avail) => console.log("NFC is " + (avail ? '' : 'not ') + 'available'));
                this.nfc.enabled().then((isOn) => console.log("NFC is " + (isOn ? '' : 'not ') + 'on'));
                this.nfc.setOnTagDiscoveredListener((data: NfcTagData) => {
                    console.log("Discovered a tag with ID " + data.id);
                }).then(() => {
                    console.log("OnTagDiscovered listener added");
                });

                this.nfc.setOnNdefDiscoveredListener((data: NfcNdefData) => {
                    console.log("Discovered a tag with ID " + data.id);
                }).then(() => {
                    console.log("OnNdef listener added");
                });

            });
        }
    }
}
```
**Note**:
`nfc.available()` and `nfc.enabled()` do return their values correctly.
**Things i've already tried**:
- Uninstalling/Installing everything.
- Trying previous versions of nativescript-nfc and nativescript.
- Using the constructor to init the Nfc class, which results the same error as: https://github.com/EddyVerbruggen/nativescript-nfc/issues/13#issuecomment-456742040

EddyVerbruggen, 2019-02-28:
Did you also try the demo app in this repo?

GijsPost, 2019-02-28:
@EddyVerbruggen I don't see how the demo has anything to do with it since it isn't Angular. Regardless, it doesn't work for me either.
```
An uncaught Exception occurred on "main" thread.
java.lang.RuntimeException: Unable to instantiate activity ComponentInfo{org.nativescript.plugindemo.nfc/com.tns.NativeScriptNfcActivity}: java.lang.ClassNotFoundException: Didn't find class "com.tns.NativeScriptNfcActivity" on path: DexPathList[[zip file "/data/app/org.nativescript.plugindemo.nfc-1/base.apk"],nativeLibraryDirectories=[/data/app/org.nativescript.plugindemo.nfc-1/lib/arm, /data/app/org.nativescript.plugindemo.nfc-1/base.apk!/lib/armeabi-v7a, /system/lib, /vendor/lib]]
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2849)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3045)
	at android.app.ActivityThread.-wrap14(ActivityThread.java)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1642)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:154)
	at android.app.ActivityThread.main(ActivityThread.java:6776)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1496)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1386)
Caused by: java.lang.ClassNotFoundException: Didn't find class "com.tns.NativeScriptNfcActivity" on path: DexPathList[[zip file "/data/app/org.nativescript.plugindemo.nfc-1/base.apk"],nativeLibraryDirectories=[/data/app/org.nativescript.plugindemo.nfc-1/lib/arm, /data/app/org.nativescript.plugindemo.nfc-1/base.apk!/lib/armeabi-v7a, /system/lib, /vendor/lib]]
	at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:380)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
	at android.app.Instrumentation.newActivity(Instrumentation.java:1086)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2839)
	... 9 more
```

EddyVerbruggen, 2019-02-28:
Please share your project and I might take a look.

GijsPost, 2019-02-28:
@EddyVerbruggen https://github.com/GijsPost/nativescript-nfc-test-temp

EddyVerbruggen, 2019-02-28:
It's not related to Angular at all; the problem with that test app is the activity needs to be replaced, not added. I've tried making that more clear in the readme: https://github.com/EddyVerbruggen/nativescript-nfc#android-setup

And you can compare your implementation with the one in the demo app:
https://github.com/EddyVerbruggen/nativescript-nfc/blob/7e811b4aae9033968fed2d513d7e208c0c707eeb/demo/app/App_Resources/Android/AndroidManifest.xml#L21-L42

closed 2019-02-28 by EddyVerbruggen

GijsPost, 2019-03-06:
Hi, sorry for the late reply.
This did indeed work. Thanks for the help!

EddyVerbruggen, 2019-03-06:
Perfect, happy to help :)

kesavanpos, 2020-05-03:
I tried too i am getting exception

kesavanpos, 2020-05-03:
Can you help me on this as I am getting this exception
": java.lang.ClassNotFoundException: Didn't find class "com.tns.NativeScriptNfcActivity"

kesavanpos, 2020-05-03:
@EddyVerbruggen Can you help me on this as I am getting this exception
": java.lang.ClassNotFoundException: Didn't find class "com.tns.NativeScriptNfcActivity"