EddyVerbruggen/nativescript-nfc #30: tns run android --bundle error with tns-android 5.2.1

femike, 2019-03-11:
Hi, i have a problem with plugin. 
I was create `tns create nfc-demo` with Angular then install plugin `tns plugin add nativescript-nfc` 
Added data to monifest and webpack.config and run `tns run android --bundle`

don't understand how import android to nfc.android.d.ts

```
ERROR in node_modules/nativescript-nfc/nfc.android.d.ts(3,18): error TS2503: Cannot find namespace 'android'.
node_modules/nativescript-nfc/nfc.android.d.ts(11,22): error TS2503: Cannot find namespace 'android'.
node_modules/nativescript-nfc/nfc.android.d.ts(12,28): error TS2503: Cannot find namespace 'android'.
node_modules/nativescript-nfc/nfc.android.d.ts(13,26): error TS2503: Cannot find namespace 'android'.
```

cwjacobs, 2019-03-12:
I have the same issue with a Typescript app using this plugin. Apparently, occurs during the "after-prepare" hook in nativescript-nfc.js. Since I don't intend to debug the plugin, it probably doesn't really matter, but it would be nice to have a clean transaction from laptop to emulator.

Great plugin, btw. Thank you.

glydetech, 2019-03-13:
I've seen this same issue with and without the --bundle flag. Try the following:

`npm install tns-platform-declarations --save-dev`

Then create a new file in the base directory of the project called references.d.ts containing:

```
/// <reference path="./node_modules/tns-platform-declarations/ios.d.ts" />
/// <reference path="./node_modules/tns-platform-declarations/android.d.ts" />
```

cwjacobs, 2019-03-13:
Worked! Thank you Andrew.

glydetech, 2019-03-13:
Maybe this should be added to the Readme file as it seems to be required.

EddyVerbruggen, 2019-03-13:
Hmm, it was not intended to be required, but if it is then I'd be happy to.

closed 2019-08-17 by EddyVerbruggen