EddyVerbruggen/nativescript-nfc #34: callback not get called when detect nfc

kousherAlam, 2019-05-15:
callback not getting called, when nfc detected. the device play the sound but nothing is happening ... 

`
import { Injectable } from "@angular/core";
import { Nfc, NfcNdefData, NfcTagData } from "nativescript-nfc";
import { Subject } from "rxjs";

@Injectable({
    providedIn: 'root'
})
export class NfcReaderService {
    nfcData = new Subject();

    private nfc: Nfc = new Nfc();
    
    constructor() {
        this.nfc.available().then((avail) => {
            this.nfc.enabled().then((on) => {
                if (on) {
                    console.log("NFC Enabled!!!");
                    this.readNFC();
                    this.onTagDiscoverd();
                } else {
                    console.log("NFC Not Enabled");
                }
            });
        });
    }

    public readNFC() {
        this.nfc.setOnNdefDiscoveredListener((data: NfcNdefData) => {
            console.log("Nfc Data: ", data);
            this.nfcData.next(data);
            if (data.message) {
                for (let m in data.message) {
                    let record = data.message[m];
                    console.log("Ndef discovered! Message record: " + record.payloadAsString);
                }
            }
        }, {
                stopAfterFirstRead: false,
                scanHint: "Scan a tag, baby!"
            }).then(() => {
                console.log("OnNdefDiscovered listener added");
            });
    }

    
    public onTagDiscoverd() {
        this.nfc.setOnTagDiscoveredListener((data: NfcTagData) => {
            console.log("Discovered a tag with ID " + data.id);
        }).then(() => {
            console.log("OnTagDiscovered listener added");
        });
    }
}

`

Ravaniss, 2019-08-28:
I have actually the same problem

EddyVerbruggen, 2019-08-30:
Please share a repo instead of a snippet and I'll take a look. Also, please post steps to reproduce this issue.

Ravaniss, 2019-08-30:
@EddyVerbruggen 
Thank you so much to see you here ! :)

I did a repo just for you <3
https://github.com/Ravaniss/nfcprob

Here is what i have did :

  1) create a project with `vue init nativescript-vue/vue-cli-template <project-name>`
      - https://nativescript-vue.org/en/docs/getting-started/quick-start/

  2) Install typescript and nfc package.

  3) copy / paste my vue code.

  4) Use `tns run --no-hmr` to run the app

  5) pass the phone on the card : I get only the ID from my doStartTagListener function. My doStartNDefListener is not call at all. ^^"

thank you !

EddyVerbruggen, 2019-08-30:
@Ravaniss This problem is on Android I guess?

What I'm seeing is when I scan an Nfc tag with your demo (after pressing all buttons) is when a tag was written to, it calls the ndef listener with all data, and if it's blank it calls the other callback.

Ravaniss, 2019-08-30:
Yes, it's on Android.

How did you get all data from the thing you scan ? O_O
Am i missing something ? Are you with IOS ?

Don't tell me i've missing something ... xD

EddyVerbruggen, 2019-08-30:
I'm testing on Android as well. Perhaps it depends on the type of tag you're scanning.. there are so many of them. But I don't see anything out of order..

Ravaniss, 2019-08-30:
Here is what i get with another app, and it's exactly these informations i need like, platform, type of tag and disponible technology

https://ibb.co/0ftrYfW

EddyVerbruggen, 2019-08-30:
Looks what I'm getting in the ndef listener. Perhaps try only setting that listener (not the tag listener).

Ravaniss, 2019-08-30:
What should i do ?

Ravaniss, 2019-08-30:
What should i do ? Because for me, my code is okay. Can you please give me an example ?

EddyVerbruggen, 2019-08-30:
Your code is fine for Android. Like I said, when only setting the ndef listener in your demo my testtags read out correctly.

Ravaniss, 2019-08-30:
Like this ? https://gist.github.com/Ravaniss/c0f7343a5b9c6978b1be01dcde5d9b39
I get this :

```
Tag discovered, but no listener was set via setOnTagDiscoveredListener. Ndef: {"id":[4,16,-114,58,41,94,-127],"techList":["android.nfc.tech.NfcA","android.nfc.tech.MifareUltralight","android.nfc.tech.NdefFormatable"]}
```

Pkurto, 2019-11-15:
I have the same problem here described. Do you discover the solution?

Ravaniss, 2019-11-15:
Nope.

Pkurto, 2019-11-15:
I am trying to use this plugin and when I run the demo code to read a tag (using my Samsung S8) I get the following error:

`Tag discovered, but no listener was set via setOnTagDiscoveredListener. Ndef: {"id":[4,25,31,-126,-39,91,-128],"techList":["android.nfc.tech.IsoDep","android.nfc.tech.NfcA","android.nfc.tech.NdefFormatable"]}`

I added a console.log in the first line of the callback function and it not appears on the logs. It seems the callback function is not called.

Any help?

Pkurto, 2019-11-18:
[UPDATE]

If I set a listener using"**setOnTagDiscoveredListener**" it will be called correctly when I scan a tag. For example:

```
OnTagDiscovered Listener set
ENTER TAG LISTENER!!!!!!!!!!!!!!!!!!!!!!!!
Tag discovered! 4,25,31,-126,-39,91,-128 android.nfc.tech.IsoDep, android.nfc.tech.NfcA, android.nfc.tech.NdefFormatable
```

Then I stopped this listener with "setOnTagDiscoveredListener(null)" and tested to set the listener for "**setOnNdefDiscoveredListener**". Then when I scan the tag again and I had the following output message:

```
doStartNdefListener Listener set
Tag discovered, but no listener was set via setOnTagDiscoveredListener. 
Ndef: {"id":[4,25,31,-126,-39,91,-128],"techList":["android.nfc.tech.IsoDep","android.nfc.tech.NfcA","android.nfc.tech.NdefFormatable"]}
```

I am trying to use the setOnNdefDiscoveredListener because it seems to have more information.
Am I using this module the wrong away?