EddyVerbruggen/nativescript-nfc #37: Android: Cannot read property getClass of undefined

rebz, 2019-06-25:
Works perfectly fine on iOS, the moment I attempt to build for android the application refuses to launch. No matter how I try to import or require your package, I get the error; `Cannot read property getClass of undefined`. I saw another github issue opened for this, but no solution was made present.

The only way I got it to work was by requiring the package within the script tag of a component. Unfortunately, that is not a solution as I need to access the Nfc on app load, before components load... that and I shouldn't have to limit myself to a component script tag to use a package.

I have been banging my head against a wall for days trying to get this to work. I am not using TypeScript and I see that your demo is... is there perhaps something you're doing with your build that is causing issues in non-TypeScript projects?

Unsure how to move forward.

EddyVerbruggen, 2019-06-26:
The plugin on npm is actually plain old JS because {N} can't run TS (it's a JS runtime).

So I don't know the exact issue, but please give version 4.0.0 a try. I've just released it. Perhaps your issue was with the custom activity that had to be configured with older versions, but that has been removed because NativeScript 5.4.0 made that obsolete.

closed 2019-06-26 by EddyVerbruggen

rebz, 2019-07-01:
@EddyVerbruggen 
I am still receiving the error after upgrading to 4.0 and updating my `AndroidManfiest` and `webpack.config` (using your updated notes in the README).

I've tried the following two methods inside of my `main.js` file:

```
const Nfc = require("nativescript-nfc").Nfc;
const handler = new Nfc();

import { Nfc } from "nativescript-nfc";
const nfc = new Nfc();
```

Below is the error I receive:
```
System.err: java.lang.RuntimeException: Unable to create application com.tns.NativeScriptApplication: com.tns.NativeScriptException: 
System.err: 
System.err: Error calling module function 
System.err: 
System.err: Error calling module function 
System.err: 
System.err: TypeError: Cannot read property 'getClass' of undefined
System.err: File: "file:///data/data/com.site.example/files/app/vendor.js, line: 612, column: 63
System.err: 
System.err: StackTrace: 
System.err:     Frame: function:'Nfc', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 612, column: 64
System.err:     Frame: function:'', file:'file:///data/data/com.site.example/files/app/bundle.js', line: 2589, column: 17
System.err:     Frame: function:'./main.js', file:'file:///data/data/com.site.example/files/app/bundle.js', line: 2641, column: 30
System.err:     Frame: function:'__webpack_require__', file:'file:///data/data/com.site.example/files/app/runtime.js', line: 751, column: 30
System.err:     Frame: function:'checkDeferredModules', file:'file:///data/data/com.site.example/files/app/runtime.js', line: 44, column: 23
System.err:     Frame: function:'webpackJsonpCallback', file:'file:///data/data/com.site.example/files/app/runtime.js', line: 31, column: 19
System.err:     Frame: function:'', file:'file:///data/data/com.site.example/files/app/bundle.js', line: 2, column: 57
System.err:     Frame: function:'require', file:'', line: 1, column: 266
System.err:     Frame: function:'', file:'file:///data/data/com.site.example/files/app/starter.js', line: 3, column: 1
System.err:     Frame: function:'require', file:'', line: 1, column: 266
System.err: 
System.err: 
System.err: TypeError: Cannot read property 'getClass' of undefined
System.err: File: "<unknown>, line: 1, column: 265
System.err: 
System.err: StackTrace: 
System.err:     Frame: function:'Nfc', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 612, column: 64
System.err:     Frame: function:'', file:'file:///data/data/com.site.example/files/app/bundle.js', line: 2589, column: 17
System.err:     Frame: function:'./main.js', file:'file:///data/data/com.site.example/files/app/bundle.js', line: 2641, column: 30
System.err:     Frame: function:'__webpack_require__', file:'file:///data/data/com.site.example/files/app/runtime.js', line: 751, column: 30
System.err:     Frame: function:'checkDeferredModules', file:'file:///data/data/com.site.example/files/app/runtime.js', line: 44, column: 23
System.err:     Frame: function:'webpackJsonpCallback', file:'file:///data/data/com.site.example/files/app/runtime.js', line: 31, column: 19
System.err:     Frame: function:'', file:'file:///data/data/com.site.example/files/app/bundle.js', line: 2, column: 57
System.err:     Frame: function:'require', file:'', line: 1, column: 266
System.err:     Frame: function:'', file:'file:///data/data/com.site.example/files/app/starter.js', line: 3, column: 1
System.err:     Frame: function:'require', file:'', line: 1, column: 266
System.err: 
System.err: 
System.err: TypeError: Cannot read property 'getClass' of undefined
System.err: File: "<unknown>, line: 1, column: 265
System.err: 
System.err: StackTrace: 
System.err:     Frame: function:'Nfc', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 612, column: 64
System.err:     Frame: function:'', file:'file:///data/data/com.site.example/files/app/bundle.js', line: 2589, column: 17
System.err:     Frame: function:'./main.js', file:'file:///data/data/com.site.example/files/app/bundle.js', line: 2641, column: 30
System.err:     Frame: function:'__webpack_require__', file:'file:///data/data/com.site.example/files/app/runtime.js', line: 751, column: 30
System.err:     Frame: function:'checkDeferredModules', file:'file:///data/data/com.site.example/files/app/runtime.js', line: 44, column: 23
System.err:     Frame: function:'webpackJsonpCallback', file:'file:///data/data/com.site.example/files/app/runtime.js', line: 31, column: 19
System.err:     Frame: function:'', file:'file:///data/data/com.site.example/files/app/bundle.js', line: 2, column: 57
System.err:     Frame: function:'require', file:'', line: 1, column: 266
System.err:     Frame: function:'', file:'file:///data/data/com.site.example/files/app/starter.js', line: 3, column: 1
System.err:     Frame: function:'require', file:'', line: 1, column: 266
System.err: 
System.err:     at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5879)
System.err:     at android.app.ActivityThread.-wrap3(ActivityThread.java)
[... 86 lines cut ...]
    "nativescript-toasty": "^1.4.0",
    "nativescript-ui-listview": "^6.3.0",
    "nativescript-ui-sidedrawer": "^4.3.0",
    "nativescript-vue": "^2.0.0",
    "nativescript-vue-devtools": "^1.2.0",
    "nativescript-vue-navigator": "0.0.3",
    "nativescript-vue-shadow": "^0.1.0",
    "tns-core-modules": "^5.4.3",
    "vuex": "^3.1.1"
  },
  "devDependencies": {
    "@babel/core": "^7.4.5",
    "@babel/preset-env": "^7.4.5",
    "babel-loader": "^8.0.6",
    "babel-traverse": "6.26.0",
    "babel-types": "6.26.0",
    "babylon": "6.18.0",
    "lazy": "1.0.11",
    "nativescript-dev-webpack": "^0.22.0",
    "nativescript-vue-template-compiler": "^2.0.0",
    "node-sass": "^4.9.2",
    "vue-loader": "^15.2.6"
  }
}
```

EddyVerbruggen, 2019-07-02:
Hi @rebz, that's because you're instantiating the plugin *before* the app starts (which is the last line in `main.js`. You could move the instantiation to a later point.

I will take a stab at hardening the plugin against this kind of usage.

rebz, 2019-07-02:
@EddyVerbruggen 

It did not matter where I loaded the plugin. Inside `main.js`, within a component's `<script>` tags, etc. The error is produced regardless of the location it is loaded.

However, I no longer receive the `getClass` error because now I get:
https://github.com/EddyVerbruggen/nativescript-nfc/issues/39

I tried downgrading to your `3.0` version, downgrading NativeScript as well. I cannot move forward with Android; always getting the foreground error. I thought maybe it was my computer, but **your demo app does not work for me either**.

I'm guessing there was some update with NativeScript which is causing this issue?

EddyVerbruggen, 2019-07-02:
I'll continue investigating in that issue. Thx.