EddyVerbruggen/nativescript-nfc #39: 4.0.0 - Foreground dispatch can only be enabled when your activity is resumed

rebz, 2019-07-01:
I've created a fresh repo for you to reproduce the bug.
https://github.com/rebz/nfc-demo-error

I am referencing a previous issue as this is where it started for me: https://github.com/EddyVerbruggen/nativescript-nfc/issues/37

With `3.0` I was able to `import` or `require` the package within the `<script>` tags of a component. However, after upgrading to `^4.0.0` I receive the following error even when pulling into the `<script>` tag. I am now unable to get the package to work using the standard NativeScript-Vue template:
`vue init nativescript-vue/vue-cli-template <project-name>`

```
JS: [Vue warn]: Error in created hook: "Error: java.lang.IllegalStateException: Foreground dispatch can only be enabled when your activity is resumed
JS:     android.nfc.NfcAdapter.enableForegroundDispatch(NfcAdapter.java:1517)
JS:     com.tns.Runtime.callJSMethodNative(Native Method)
JS:     com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1203)
JS:     com.tns.Runtime.callJSMethodImpl(Runtime.java:1083)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:1070)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:1050)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:1042)
JS:     com.tns.NativeScriptActivity.onCreate(NativeScriptActivity.java:19)
JS:     android.app.Activity.performCreate(Activity.java:6942)
JS:     android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1126)
JS:     android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2880)
JS:     android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2988)
JS:     android.app.ActivityThread.-wrap14(ActivityThread.java)
JS:     android.app.ActivityThread$H.handleMessa...
System.err: java.lang.RuntimeException: Unable to start activity ComponentInfo{com.site.example/com.tns.NativeScriptActivity}: com.tns.NativeScriptException: 
System.err: Calling js method onCreate failed
System.err: 
System.err: Error: java.lang.IllegalStateException: Foreground dispatch can only be enabled when your activity is resumed
System.err:     android.nfc.NfcAdapter.enableForegroundDispatch(NfcAdapter.java:1517)
System.err:     com.tns.Runtime.callJSMethodNative(Native Method)
System.err:     com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1203)
System.err:     com.tns.Runtime.callJSMethodImpl(Runtime.java:1083)
System.err:     com.tns.Runtime.callJSMethod(Runtime.java:1070)
System.err:     com.tns.Runtime.callJSMethod(Runtime.java:1050)
System.err:     com.tns.Runtime.callJSMethod(Runtime.java:1042)
System.err:     com.tns.NativeScriptActivity.onCreate(NativeScriptActivity.java:19)
System.err:     android.app.Activity.performCreate(Activity.java:6942)
System.err:     android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1126)
System.err:     android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2880)
System.err:     android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2988)
System.err:     android.app.ActivityThread.-wrap14(ActivityThread.java)
System.err:     android.app.ActivityThread$H.handleMessage(ActivityThread.java:1631)
System.err:     android.os.Handler.dispatchMessage(Handler.java:102)
System.err:     android.os.Looper.loop(Looper.java:154)
System.err:     android.app.ActivityThread.main(ActivityThread.java:6682)
System.err:     java.lang.reflect.Method.invoke(Native Method)
System.err:     com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1534)
System.err:     com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1424)
System.err: File: "file:///data/data/com.site.example/files/app/vendor.js, line: 9670, column: 6
System.err: 
System.err: StackTrace: 
System.err:     Frame: function:'Nfc', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 462, column: 27
System.err:     Frame: function:'created', file:'file:///data/data/com.site.example/files/app/bundle.js', line: 421, column: 17
System.err:     Frame: function:'callHook', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 10735, column: 21
System.err:     Frame: function:'Vue._init', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 18651, column: 5
System.err:     Frame: function:'VueComponent', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 18802, column: 12
System.err:     Frame: function:'createComponentInstanceForVnode', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 18335, column: 10
System.err:     Frame: function:'init', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 18182, column: 45
System.err:     Frame: function:'createComponent', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 10911, column: 9
System.err:     Frame: function:'createElm', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 10860, column: 9
System.err:     Frame: function:'createChildren', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 10998, column: 9
System.err:     Frame: function:'createElm', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 10883, column: 9
System.err:     Frame: function:'patch', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 11468, column: 7
System.err:     Frame: function:'Vue._update', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 10481, column: 19
System.err:     Frame: function:'updateComponent', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 10590, column: 10
System.err:     Frame: function:'get', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 9665, column: 25
System.err:     Frame: function:'Watcher', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 9652, column: 45
System.err:     Frame: function:'mountComponent', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 10596, column: 3
System.err:     Frame: function:'push.../node_modules/nativescript-vue/dist/index.js.Vue.$mount', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 19189, column: 10
System.err:     Frame: function:'init', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 18183, column: 13
[... 151 lines cut ...]
System.err:     Frame: function:'mountComponent', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 10596, column: 3
System.err:     Frame: function:'push.../node_modules/nativescript-vue/dist/index.js.Vue.$mount', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 19189, column: 10
System.err:     Frame: function:'', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 19205, column: 10
System.err:     Frame: function:'Observable.notify', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 23740, column: 15
System.err:     Frame: function:'notifyLaunch', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 46149, column: 15
System.err:     Frame: function:'ActivityCallbacksImplementation.setActivityContent', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 46072, column: 20
System.err:     Frame: function:'ActivityCallbacksImplementation.onCreate', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 45877, column: 10
System.err:     Frame: function:'NativeScriptActivity.onCreate', file:'file:///data/data/com.site.example/files/app/vendor.js', line: 43104, column: 21
System.err: 
System.err:     at com.tns.Runtime.callJSMethodNative(Native Method)
System.err:     at com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1203)
System.err:     at com.tns.Runtime.callJSMethodImpl(Runtime.java:1083)
System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:1070)
System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:1050)
System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:1042)
System.err:     at com.tns.NativeScriptActivity.onCreate(NativeScriptActivity.java:19)
System.err:     at android.app.Activity.performCreate(Activity.java:6942)
System.err:     at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1126)
System.err:     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2880)
System.err:     ... 9 more
System.err: Caused by: java.lang.IllegalStateException: Foreground dispatch can only be enabled when your activity is resumed
System.err:     at android.nfc.NfcAdapter.enableForegroundDispatch(NfcAdapter.java:1517)
System.err:     ... 19 more
Successfully synced application com.site.example on device 0915f9d4542e2a03.
```

rebz, 2019-07-01:
I downgraded to `3.0.3` but am still receiving this issue. Might be something else going on. I saw another issue thread, now closed, with this error but no solution other than a new version release of `2.1.2`.

EddyVerbruggen, 2019-07-02:
Let me check, thanks for the details and demo app! I'm sure that will help.

EddyVerbruggen, 2019-07-02:
Please give 4.0.1 a go. I should fix this issue (tested with your demo app as well).

closed 2019-07-02 by EddyVerbruggen

rebz, 2019-07-02:
@EddyVerbruggen - I can confirm that `4.0.1` is working in the repo I linked above. I'm heading out for the evening but I will do some further testing this week as I port it over to my app.

Genuinely appreciate you taking the time to work on this.

rebz, 2019-07-02:
Took a bit of uninstalling and re-install to clear cache and tweak all files. It is now working in my main app.

Thank you!

EddyVerbruggen, 2019-07-03:
Awesome, thx for the feedback!