EddyVerbruggen/nativescript-nfc #40: Error: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.nfc.Tag.hasTech(int)' on a null object reference

larscifi, 2019-07-03:
Hi, when i try to _write_ an nfc tag on android, the following error occurs:

I´m working with the latest version

` [Error: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.nfc.Tag.hasTech(int)' on a null object reference
JS:     android.nfc.tech.Ndef.get(Ndef.java:146)
JS:     com.tns.Runtime.callJSMethodNative(Native Method)
JS:     com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1203)
JS:     com.tns.Runtime.callJSMethodImpl(Runtime.java:1083)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:1070)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:1050)
JS:     com.tns.Runtime.callJSMethod(Runtime.java:1042)
JS:     com.tns.gen.org.nativescript.widgets.Async_CompleteCallback.onComplete(Async_CompleteCallback.java:20)
JS:     org.nativescript.widgets.Async$Http$HttpRequestTask.onPostExecute(Async.java:573)
JS:     org.nativescript.widgets.Async$Http$1$1.run(Async.java:304)
JS:     android.os.Handler.handleCallback(Handler.java:789)
JS:     android.os.Handler.dispatchMessage(Handler.java:98)
JS:     android.os.Looper.loop(Looper.java:180)
JS:     android.app.ActivityThread.main(ActivityThread.java:6950)
JS:     java.lang.reflect.Method.invo...`

a-backman, 2019-08-16:
Any progress on this issue? I get the same error.

Fatfrido, 2019-08-16:
I've got the same problem with Nativescript 6.0.0.

EddyVerbruggen, 2019-08-17:
Interestingly I can't reproduce this.. how do you trigger it exactly?

Looking at the stacktrace you could try and guard [this line](https://github.com/EddyVerbruggen/nativescript-nfc/blob/a8d18da75261d3a8b87352bbc5d9a38ade671394/src/nfc.android.ts#L41); only run that bit of code if `tag` is not null. You can go into `node_modules/nativescript-nfc` and give it a go.

EddyVerbruggen, 2019-08-17:
I'm just going to assume the aforementioned issue is the problem on your end, and publish 4.0.2 in a bit.

Fatfrido, 2019-08-17:
So.. I quickly created a new project just to test the writing [here](https://github.com/Fatfrido/ns-nfc-test) (when an empty tag is discovered it should write a string to it). However, I still get this error. 
I've also tried to guard the line @EddyVerbruggen mentioned earlier (thanks for your input, btw), but this part of the code is not executed, since I only try to write on empty tags, I guess.
I used Nativescript 6.0.0 and tested on Android 9.

EddyVerbruggen, 2019-08-17:
@Fatfrido Thx for the repo. Indeed, when writing a tag I can reproduce this error.

EddyVerbruggen, 2019-08-17:
@Fatfrido Should be much better with 4.0.3.

closed 2019-08-17 by EddyVerbruggen

Fatfrido, 2019-08-17:
@EddyVerbruggen thanks for the quick fix mate. You are awesome :D

a-backman, 2019-08-18:
@EddyVerbruggen thanks for the fix!