EddyVerbruggen/nativescript-nfc #41: What is the current / correct build procedure for the demo?

samarqhand, 2019-07-10:
I cannot get the demo to run properly, I have tried `npm run demo.android`, `tns run android` and `tns build android`

Both methods seem to work, but produce an error either in the emulator.

my `tns info`  -

> Component nativescript has 5.4.2 version and is up to date. 
> Component tns-core-modules has 6.0.0-next-2019-06-20-155941-01 version and is up to date. 
> Component tns-android has 6.0.0-2019-06-11-172137-01 version and is up to date. 
> Component tns-ios has 6.0.0-2019-06-10-154118-03 version and is up to date

And the error in the emulator

> An uncaught Exception occurred on "main" thread.
> java.lang.RuntimeException: Unable to create application com.tns.NativeScriptApplication: com.tns.NativeScriptException: Application entry point file not found. Please specify the file in package.json otherwise make sure the file index.js or bootstrap.js exists.\nIf using typescript make sure your entry point file is transpiled to javascript.
> 	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5876)
> 	at android.app.ActivityThread.access$1100(ActivityThread.java:199)
> 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1650)
> 	at android.os.Handler.dispatchMessage(Handler.java:106)
> 	at android.os.Looper.loop(Looper.java:193)
> 	at android.app.ActivityThread.main(ActivityThread.java:6669)
> 	at java.lang.reflect.Method.invoke(Native Method)
> 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
> 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
> Caused by: com.tns.NativeScriptException: Application entry point file not found. Please specify the file in package.json otherwise make sure the file index.js or bootstrap.js exists.\nIf using typescript make sure your entry point file is transpiled to javascript.
> 	at com.tns.Module.bootstrapApp(Module.java:311)
> 	at com.tns.Runtime.run(Runtime.java:624)
> 	at com.tns.NativeScriptApplication.onCreate(NativeScriptApplication.java:21)
> 	at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1154)
> 	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5871)
> 	... 8 more
> Caused by: com.tns.NativeScriptException: Failed to find module: "./", relative to: app//
> 	at com.tns.Module.resolvePathHelper(Module.java:146)
> 	at com.tns.Module.bootstrapApp(Module.java:309)
> 	... 12 more

EddyVerbruggen, 2019-07-10:
I'm testing this plugin with NativeScript 6, so try with the NativeScript 6 RC: `npm -g nativescript@rc`. And to go back to 5.2 (or whatever you're running` do `npm -g nativescript@5.2`.

zhiyong80, 2019-07-22:
With NativeScript 6, somehow I need to change the following line in app/app.ts:
```
application.start({ moduleName: 'main-page' });
```
to
```
application._start({ moduleName: 'main-page' });
```
Otherwise I get the following error message:

TS2551: Property 'start' does not exist on type 'typeof import("nativescript-nfc-master/demo/node_modules/tns-core-modules/application/application")'. Did you mean '_start'?
Webpack compilation complete.
Executing webpack failed with exit code 2.

EddyVerbruggen, 2019-07-22:
I’ll update that demo for N6 one day. Btw, start is now run.

EddyVerbruggen, 2019-08-17:
Demo has been updated to N6.

closed 2019-08-17 by EddyVerbruggen