EddyVerbruggen/nativescript-nfc #44: Error UnhandledPromiseRejectionWarning and NFC permission required

furkarayel, 2019-08-02:
Hello. I asked a question on stackoverflow where I got told I should open an issue on github, sorry for not opening an issue here. 
I'm currently working on a project I have to read/write NFC tags with a Samsung Tablet which supports NFC.

I simply created a new project based on Vue.Js with NativeScript Sidekick and installed the nativescript-nfc. 

I first of all implemented a method which executes the available function of the plugin, which returns 'yes'. After that I'm getting the error that the NFC permission is required. 

I tried to change the Manifest file to let the user ask for the permissions but somehow I couldn't make it work. (Most probably the reason will be that it's not possible on LiveSync? I'm new to NativeScript sorry.) 

I get these logs on my device console:

> LOG from device Galaxy Tab Active2: '\'NativeScript-Vue has "Vue.config.silent" set to true, to see output logs set it to false.\''
> LOG from device Galaxy Tab Active2: '\'Yes\''
> LOG from device Galaxy Tab Active2: An uncaught Exception occurred on "main" thread.
> Calling js method run failed
> Error: java.lang.SecurityException: NFC permission required: Neither user 10155 nor current process has android.permission.NFC.
>     android.os.Parcel.readException(Parcel.java:2029)
>     android.os.Parcel.readException(Parcel.java:1975)
>     android.nfc.INfcAdapter$Stub$Proxy.setForegroundDispatch(INfcAdapter.java:1012)
>     android.nfc.NfcAdapter.enableForegroundDispatch(NfcAdapter.java:1648)
>     com.tns.Runtime.callJSMethodNative(Native Method)
>     com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1242)
>     com.tns.Runtime.callJSMethodImpl(Runtime.java:1122)
>     com.tns.Runtime.callJSMethod(Runtime.java:1109)
>     com.tns.Runtime.callJSMethod(Runtime.java:1089)
>     com.tns.Runtime.callJSMethod(Runtime.java:1081)
>     com.tns.gen.java.lang.Runnable.run(Runnable.java:17)
>     android.os.Handler.handleCallback(Handler.java:790)
>     android.os.Handler.dispatchMessage(Handler.java:99)
>     android.os.Looper.loop(Looper.java:164)
>     android.app.ActivityThread.main(ActivityThread.java:7000)
>     java.lang.reflect.Method.invoke(Native Method)
>     com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:441)
>     com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1408)
> 
> StackTrace:
> 	Frame: function:'', file:'file:///data/data/org.nativescript.preview/files/app/vendor.js', line: 672, column: 31
> 	Frame: function:'invoke', file:'file:///data/data/org.nativescript.preview/files/app/tns_modules/tns-core-modules/timer/timer.js', line: 19, column: 45
> 	Frame: function:'run', file:'file:///data/data/org.nativescript.preview/files/app/tns_modules/tns-core-modules/timer/timer.js', line: 23, column: 7
> 	at com.tns.Runtime.callJSMethodNative(Native Method)
> 	at com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1242)
> 	at com.tns.Runtime.callJSMethodImpl(Runtime.java:1122)
> 	at com.tns.Runtime.callJSMethod(Runtime.java:1109)
> 	at com.tns.Runtime.callJSMethod(Runtime.java:1089)
> 	at com.tns.Runtime.callJSMethod(Runtime.java:1081)
> 	at com.tns.gen.java.lang.Runnable.run(Runnable.java:17)
> 	at android.os.Handler.handleCallback(Handler.java:790)
> 	at android.os.Handler.dispatchMessage(Handler.java:99)
> 	at android.os.Looper.loop(Looper.java:164)
> 	at android.app.ActivityThread.main(ActivityThread.java:7000)
> 	at java.lang.reflect.Method.invoke(Native Method)
> 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:441)
> 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1408)
> Caused by: java.lang.SecurityException: NFC permission required: Neither user 10155 nor current process has android.permission.NFC.
> 	at android.os.Parcel.readException(Parcel.java:2029)
> 	at android.os.Parcel.readException(Parcel.java:1975)
> 	at android.nfc.INfcAdapter$Stub$Proxy.setForegroundDispatch(INfcAdapter.java:1012)
> 	at android.nfc.NfcAdapter.enableForegroundDispatch(NfcAdapter.java:1648)
> 	... 14 more

and these errors are on the errors section:

> [19-08-02 15:27:52.337] (CLI) (node:18276) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'hash' of null
>     at PreviewAppController.<anonymous> (C:\Users\XXXXXXXX\AppData\Roaming\npm\node_modules\nativescript\lib\controllers\preview-app-controller.js:130:64)
>     at Generator.next (<anonymous>)
>     at fulfilled (C:\Users\XXXXXXXX\AppData\Roaming\npm\node_modules\nativescript\lib\controllers\preview-app-controller.js:10:58)
>     at processTicksAndRejections (internal/process/next_tick.js:81:5)
>     at emitWarning (internal/process/promises.js:99:15)
>     at processPromiseRejections (internal/process/promises.js:140:9)
>     at processTicksAndRejections (internal/process/next_tick.js:82:32)
> [19-08-02 15:27:52.340] (CLI) (node:18276) TypeError: Cannot read property 'hash' of null
>     at PreviewAppController.<anonymous> (C:\Users\XXXXXXXX\AppData\Roaming\npm\node_modules\nativescript\lib\controllers\preview-app-controller.js:130:64)
>     at Generator.next (<anonymous>)
>     at fulfilled (C:\Users\XXXXXXXX\AppData\Roaming\npm\node_modules\nativescript\lib\controllers\preview-app-controller.js:10:58)
>     at processTicksAndRejections (internal/process/next_tick.js:81:5)
> (node:18276) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
>     at emitDeprecationWarning (internal/process/promises.js:113:13)
>     at emitWarning (internal/process/promises.js:106:3)
>     at processPromiseRejections (internal/process/promises.js:140:9)
>     at processTicksAndRejections (internal/process/next_tick.js:82:32)
> [19-08-02 15:32:26.783] (CLI) (node:18276) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'hash' of null
>     at PreviewAppController.<anonymous> (C:\Users\XXXXXXXX\AppData\Roaming\npm\node_modules\nativescript\lib\controllers\preview-app-controller.js:130:64)
>     at Generator.next (<anonymous>)
>     at fulfilled (C:\Users\XXXXXXXX\AppData\Roaming\npm\node_modules\nativescript\lib\controllers\preview-app-controller.js:10:58)
>     at processTicksAndRejections (internal/process/next_tick.js:81:5)
>     at emitWarning (internal/process/promises.js:99:15)
>     at processPromiseRejections (internal/process/promises.js:140:9)
>     at processTicksAndRejections (internal/process/next_tick.js:82:32)
> (node:18276) TypeError: Cannot read property 'hash' of null
>     at PreviewAppController.<anonymous> (C:\Users\XXXXXXXX\AppData\Roaming\npm\node_modules\nativescript\lib\controllers\preview-app-controller.js:130:64)
>     at Generator.next (<anonymous>)
>     at fulfilled (C:\Users\XXXXXXXX\AppData\Roaming\npm\node_modules\nativescript\lib\controllers\preview-app-controller.js:10:58)
>     at processTicksAndRejections (internal/process/next_tick.js:81:5)
> [19-08-02 15:32:57.564] (CLI) (node:18276) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'hash' of null
>     at PreviewAppController.<anonymous> (C:\Users\XXXXXXXX\AppData\Roaming\npm\node_modules\nativescript\lib\controllers\preview-app-controller.js:130:64)
>     at Generator.next (<anonymous>)
>     at fulfilled (C:\Users\XXXXXXXX\AppData\Roaming\npm\node_modules\nativescript\lib\controllers\preview-app-controller.js:10:58)
>     at processTicksAndRejections (internal/process/next_tick.js:81:5)
>     at emitWarning (internal/process/promises.js:99:15)
>     at processPromiseRejections (internal/process/promises.js:140:9)
>     at processTicksAndRejections (internal/process/next_tick.js:82:32)
> (node:18276) TypeError: Cannot read property 'hash' of null
>     at PreviewAppController.<anonymous> (C:\Users\XXXXXXXX\AppData\Roaming\npm\node_modules\nativescript\lib\controllers\preview-app-controller.js:130:64)
>     at Generator.next (<anonymous>)
>     at fulfilled (C:\Users\XXXXXXXX\AppData\Roaming\npm\node_modules\nativescript\lib\controllers\preview-app-controller.js:10:58)
>     at processTicksAndRejections (internal/process/next_tick.js:81:5)

I would be really glad if you guys could help me solving these issues. Thank you in advance.

EddyVerbruggen, 2019-08-02:
Do you use the Preview app? That's a wrapper app which only ships a certain amount of plugins, this one not being one of those. So if you want to test NFC you'll need to build your own app, and the plugin will handle adding those permissions for you.

furkarayel, 2019-08-05:
It worked with a local build without errors. Love you man.

closed 2019-08-17 by EddyVerbruggen