EddyVerbruggen/nativescript-nfc #51: Cannot find namespace 'android'

Rav-Naz, 2019-11-06:
after type in console `tns run` I'm getting this error:

```
ERROR in node_modules/nativescript-nfc/nfc.android.d.ts:3:18 - error TS2503: Cannot find namespace 'android'.

3     savedIntent: android.content.Intent;
                   ~~~~~~~
node_modules/nativescript-nfc/nfc.android.d.ts:11:22 - error TS2503: Cannot find namespace 'android'.

11     ndefToJSON(ndef: android.nfc.tech.Ndef): NfcNdefData;
                        ~~~~~~~
node_modules/nativescript-nfc/nfc.android.d.ts:12:28 - error TS2503: Cannot find namespace 'android'.

12     messageToJSON(message: android.nfc.NdefMessage): Array<NfcNdefRecord>;
                              ~~~~~~~
node_modules/nativescript-nfc/nfc.android.d.ts:13:26 - error TS2503: Cannot find namespace 'android'.

13     recordToJSON(record: android.nfc.NdefRecord): NfcNdefRecord;
                            ~~~~~~~
```

I'm working on Android Emulator, in Nativescript Angular

EddyVerbruggen, 2019-11-06:
Thx for reporting. There are a few ways around this, but I think it's best to fix that in the plugin itself - give 4.1.0 a spin.

closed 2019-11-06 by EddyVerbruggen

Rav-Naz, 2019-11-06:
I'm fixed it by install tns-platform-declarations and added reference.d.ts
on root folder. I'm suggest that you should add this  infromation in
readme.md.

EddyVerbruggen, 2019-11-06:
That was what I could have said, but I did one better and fixed it in the plugin like I mentioned.

Rav-Naz, 2019-11-08:
Hey, is there any way to enable NFC only in one component? Already it is
enabled on whole application, and everytime I'm getting this information:

*Ndef discovered, but no listener was set via setOnNdefDiscoveredListener. *

I'm wish to use nfc only in components where I have listeners.
Thanks for help in advance.
Best regards, Rafał

EddyVerbruggen, 2019-11-08:
We can remove that console.log, if that helps. But please don't hijack other issues with unrelated questions.