EddyVerbruggen/nativescript-nfc #52: Add writeTag support for iOS

gasparrobi, 2019-12-26:
writing to a tag is now supported in iOS 13.

see `writeNDEF` at [https://developer.apple.com/documentation/corenfc/nfcndeftag](https://developer.apple.com/documentation/corenfc/nfcndeftag)

EddyVerbruggen, 2019-12-27:
Oh cool!

EddyVerbruggen, 2019-12-30:
I've added code to do a little PoC of this feature, but in my tests (on line 165) `nfcNdefTag.writeNDEFCompletionHandler` is `undefined`.

Here's the (commented) code.. feel free to play with it:

https://github.com/EddyVerbruggen/nativescript-nfc/blob/d4414b3a55f7d8eff6ed839ef531cb0262b10ad3/src/nfc.ios.ts#L150-L171

gasparrobi, 2020-01-06:
I'm trying to use your snippet. While the `readerSessionDidDetectTags` is called, the `tags` input parameter is undefined. I'm not sure why. I've tried with both empty and non-empty tags

joeelia, 2020-01-13:
Looking forward to this working

sashok1337, 2020-02-14:
Maybe it's somehow connected to NativeScript/ios-runtime#1251 ?  There is also native object transforms to JS without needed methods.

demetrius-tech, 2020-05-08:
Do you guys need any help? I need this feature and may be you've already has some progress or I can start working on this.

joeelia, 2020-05-08:
No progress. iOS wrote NFC still dosent work for me

demetrius-tech, 2020-05-08:
Check it out: https://github.com/chariotsolutions/phonegap-nfc 

The problem is that beginSession() is used instead of scanNdef(). 

"On iOS, the user must start a NFC session to scan for a tag. This is different from Android which can constantly scan for NFC tags. The nfc.scanNdef and nfc.scanTag functions start a NFC scanning session. The NFC tag is returned to the caller via a Promise. If your existing code uses the deprecated nfc.beginSession, update it to use nfc.scanNdef."

They recently fixed this on phonegap-nfc: https://github.com/chariotsolutions/phonegap-nfc/commit/a3b1e3ad5c11267aa3a8468519b4bb7e4d84800a

EddyVerbruggen, 2020-05-08:
A PR would be very much appreciated!

demetrius-tech, 2020-05-08:
Sure, but it will take time as I don't have the dev env set up for iOS. I am developing for Android now. But I am happy to help. I've never done a PR in my life, but I am eager to learn :D

joeelia, 2020-05-23:
Perhaps we have to call **queryNDEFStatus** before trying **writeNDEF** as per https://developer.apple.com/documentation/corenfc/nfcndeftag/3075574-writendef?language=objc. Found this native Swift sample app code for iOS

```
    func readerSession(_ session: NFCNDEFReaderSession, didDetect tags: [NFCNDEFTag]) {
        if tags.count > 1 {
            session.alertMessage = "More than 1 tags found. Please present only 1 tag."
            self.tagRemovalDetect(tags.first!)
            return
        }
        
        // You connect to the desired tag.
        let tag = tags.first!
        session.connect(to: tag) { (error: Error?) in
            if error != nil {
                session.restartPolling()
                return
            }
            
            // You then query the NDEF status of tag.
            tag.queryNDEFStatus() { (status: NFCNDEFStatus, capacity: Int, error: Error?) in
                if error != nil {
                    session.invalidate(errorMessage: "Fail to determine NDEF status.  Please try again.")
                    return
                }
                
                if status == .readOnly {
                    session.invalidate(errorMessage: "Tag is not writable.")
                } else if status == .readWrite {
                    if self.ndefMessage!.length > capacity {
                        session.invalidate(errorMessage: "Tag capacity is too small.  Minimum size requirement is \(self.ndefMessage!.length) bytes.")
                        return
                    }
                    
                    // When a tag is read-writable and has sufficient capacity,
                    // write an NDEF message to it.
                    tag.writeNDEF(self.ndefMessage!) { (error: Error?) in
                        if error != nil {
                            session.invalidate(errorMessage: "Update tag failed. Please try again.")
                        } else {
                            session.alertMessage = "Update success!"
                            session.invalidate()
                        }
                    }
                } else {
                    session.invalidate(errorMessage: "Tag is not NDEF formatted.")
                }
            }
        }
    }
```

But I dont know how to implement this in Nativescript :D

demetrius-tech, 2020-06-23:
@EddyVerbruggen: I've done good progress on implementing writeTag for iOS, it actually works, please check here: https://github.com/demetrius-tech/nativescript-nfc/blob/dev-uid/src/nfc.ios.ts 

Kindly support me in finding a solution for getting the UID #55 I'm pretty desperate.

timh5, 2021-12-04:
@demetrius-tech  - what is the status of this feature? it seems that we'll have to add some obj-c code to make it work?
Is that the latest development? Can I hire you to consult on some realtime questions as we try to impliment this?
Please email me your skype or google/chat info. timhon@gmail.com

demetrius-tech, 2021-12-04:
@timh5 it was working in the past, with version 7. Let me check if I committed the code in my repo and I will support you for free, as I don't have much time for other projects.