EddyVerbruggen/nativescript-nfc #53: System.err: Error: java.lang.IllegalStateException: Foreground dispatch can only be enabled when your activity is resumed

cloudhx, 2020-01-15:
Try to `tns run android` to an Android 9 device, the app will check the availability of the nfc at startup as shown below.

```typescript
checkNFC() {
    this.nfc = new Nfc();
    if (!this.isDoneChecking) {
      this.showLoader();
      Promise.all([this.nfc.available(), this.nfc.enabled()]).then(values => {
        this.isNfcAvailable = values[0];
        this.isNfcEnabled = values[1];
        this.loader.hide();
        this.isDoneChecking = true;
      });
    }
  }
```

If the device is unlocked when the app is starting, it works well. If the device is locked when starting the app, the following errors are logged in the console.

```
System.err: An uncaught Exception occurred on "main" thread.
System.err: Calling js method run failed
System.err: Error: java.lang.IllegalStateException: Foreground dispatch can only be enabled when your activity is resumed
System.err: 
System.err: StackTrace:
System.err: push.../node_modules/nativescript-nfc/nfc.js.Nfc.initNfcAdapter(file:///node_modules/nativescript-nfc/nfc.js:368:0)
System.err:     at Nfc(file:///node_modules/nativescript-nfc/nfc.js:231:0)
System.err:     at module.exports.push../shared/services/nfc.service.ts.NfcService.checkNFC(file:///app/shared/services/nfc.service.ts:18:15)
System.err:     at (file:///app/home/home.component.ts:52:22)
System.err:     at push.../node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invokeTask(file:///node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js:421:0)
System.err:     at onInvokeTask(file:///node_modules/@angular/core/fesm5/core.js:26247:0)
System.err:     at push.../node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invokeTask(file:///node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js:420:0)
System.err:     at push.../node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js.Zone.runTask(file:///node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js:188:0)
System.err:     at push.../node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js.ZoneTask.invokeTask(file:///node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js:496:0)
System.err:     at ZoneTask.invoke(file:///node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js:485:0)
System.err:     at timer(file:///node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js:1561:0)
System.err:     at invoke(file:///node_modules/@nativescript/core/timer/timer.js:20:30)
System.err:     at push.../node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invoke(file:///node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js:388:0)
System.err:     at onInvoke(file:///node_modules/@angular/core/fesm5/core.js:26256:0)
System.err:     at push.../node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js.ZoneDelegate.invoke(file:///node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js:387:0)
System.err:     at push.../node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js.Zone.runGuarded(file:///node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js:151:0)
System.err:     at (file:///node_modules/@nativescript/angular/zone-js/dist/zone-nativescript.js:129:0)
System.err:     at run(file:///node_modules/@nativescript/core/timer/timer.js:24:0)
System.err:     at com.tns.Runtime.callJSMethodNative(Native Method)
System.err:     at com.tns.Runtime.dispatchCallJSMethodNative(Runtime.java:1286)
System.err:     at com.tns.Runtime.callJSMethodImpl(Runtime.java:1173)
System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:1160)
System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:1138)
System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:1134)
System.err:     at com.tns.gen.java.lang.Runnable.run(Runnable.java:17)
System.err:     at android.os.Handler.handleCallback(Handler.java:873)
System.err:     at android.os.Handler.dispatchMessage(Handler.java:99)
System.err:     at android.os.Looper.loop(Looper.java:193)
System.err:     at android.app.ActivityThread.main(ActivityThread.java:6715)
System.err:     at java.lang.reflect.Method.invoke(Native Method)
System.err:     at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:911)
System.err: Caused by: java.lang.IllegalStateException: Foreground dispatch can only be enabled when your activity is resumed
System.err:     at android.nfc.NfcAdapter.enableForegroundDispatch(NfcAdapter.java:1352)
System.err:     ... 14 more
```

After some debugging, it turns out that the error occurred during Nfc instantiating: `this.nfc = new Nfc();`

I don't know if this is a known issue or something that can be fixed.

Thanks!

PavelSuk98, 2020-05-31:
Any news on this bug, i am now working with NFC and managed to found that if app is running and it is found on taskbar when you are selecting from other apps.. then it is working but if you wont find it then error occurs.. 

Any ideas how to fix it ?

e-macs, 2020-08-27:
Any update on this issue? I need 4.1.0 version to be able to write tags, downgrade is not an option for me...