EddyVerbruggen/nativescript-nfc #55: iOS: How get tag uid?

Atomico001, 2020-04-29:
NfcNdefData should contain the id. In android it work but in iOS it is empty.

Is it possible get the tag uid?

Puschor, 2020-05-17:
I have the same problem. Is there any workaround available? This is what I found so far regarding this issue:

https://stackoverflow.com/questions/57978507/reading-uids-of-nfc-cards-in-ios-13
https://stackoverflow.com/questions/51822226/corenfc-not-reading-uid-in-ios

Is this maybe related to iOS 13?

I tried to read the tag with a different iOS app and it properly shows the uid.

demetrius-tech, 2020-06-18:
I started to work on this inspired by the phonegap-nfc plugin: https://github.com/chariotsolutions/phonegap-nfc/blob/master/src/ios/NfcPlugin.m

But, being a newbie in NativeScript, I have some issues and I need some help, please.
I cannot access some properties from the NFCTag objects. 

The code can be found here and I have commented what issues I have:
https://github.com/demetrius-tech/nativescript-nfc/tree/dev

I am using demo-ng app.

Plugin iOS file is here: https://github.com/demetrius-tech/nativescript-nfc/blob/dev/src/nfc.ios.ts

Check tagReaderSessionDidDetectTags() and getTagUID() methods.

demetrius-tech, 2020-06-23:
Does anyone know how to translate this Objective C code to TypeScript?

id<NFCTag> tag = [tags firstObject];
NSMutableDictionary *tagMetaData = [self getTagInfo:tag];

Simply doing: const tag = tags[0]; does not return anything in tag.type

Here is what i've done so far: https://github.com/demetrius-tech/nativescript-nfc/blob/dev-uid/src/nfc.ios.ts (tagReaderSessionDidDetectTags() method)

EddyVerbruggen, 2020-06-23:
@demetrius-tech haveyour tried this? `const tag = tags.firstObject`

demetrius-tech, 2020-06-23:
firstObject only works if I do like this: const tag = NSArray.arrayWithArray(tags).firstObject; otherwise it is not available. 

Even so, if I do console.log(tag.type) it says "undefined".

demetrius-tech, 2020-06-23:
Found a way to get the type, I will just compare this:

if (NFCTag.prototype.asNFCMiFareTag.apply(tag) === tag) {
      console.log("THIS IS A MiFare tag");
      tag.type = NFCTagType.MiFare;
      let mifareTag:NFCMiFareTag = NFCTag.prototype.asNFCMiFareTag.apply(tag);
      uid = NSData.alloc().initWithData(mifareTag.identifier);
    } else if (NFCTag.prototype.asNFCISO15693Tag.apply(tag) === tag) {
      console.log("ISO15693Tag");
    } else if (NFCTag.prototype.asNFCISO7816Tag.apply(tag) === tag) {
      console.log("NFCISO7816Tag");
    } else if (NFCTag.prototype.asNFCFeliCaTag.apply(tag) === tag) {
      console.log("FeliCaTag")
    } else {
      console.log("Unknown Tag");
    }

Next step is to extract the identifier.

demetrius-tech, 2020-06-23:
Getting the UID on iOS using NativeScript is not trivial. It always displays empty. Any ideas?

My code: https://github.com/demetrius-tech/nativescript-nfc/blob/dev-uid/src/nfc.ios.ts (getTagUID method)

Related StackOverflow topics:
https://stackoverflow.com/questions/46504035/little-endian-byte-order-ios-ble-scan-response
https://stackoverflow.com/questions/46518084/nativescript-get-string-from-interop-reference

demetrius-tech, 2020-06-25:
I've managed to get the tag UID by adding native code ObjC library. 

You can check here: https://github.com/demetrius-tech/nativescript-nfc/tree/dev-ios-write-uid/src

Also, you can find there the code for writing tags for iOS. 

If interested in PR, please let me know.

Atomico001, 2020-07-07:
Hello demetrius, so you succesfull? can you explain how you did and where i retreive the tag id?

> 
> 
> I've managed to get the tag UID by adding native code ObjC library.
> 
> You can check here: https://github.com/demetrius-tech/nativescript-nfc/tree/dev-ios-write-uid/src
> 
> Also, you can find there the code for writing tags for iOS.
> 
> If interested in PR, please let me know.

demetrius-tech, 2020-07-13:
@Atomico001: you can find the ObjC code here: https://github.com/demetrius-tech/nativescript-nfc/tree/dev-ios-write-uid/src/platforms/ios/xcode/NfcNativeHelper

Created the libNfcNativeHelper.a library and added it to the src/platforms/ios (together with a .d.ts spec) : https://github.com/demetrius-tech/nativescript-nfc/tree/dev-ios-write-uid/src/platforms/ios

Atomico001, 2020-07-13:
> @Atomico001: you can find the ObjC code here: https://github.com/demetrius-tech/nativescript-nfc/tree/dev-ios-write-uid/src/platforms/ios/xcode/NfcNativeHelper
>
> Created the libNfcNativeHelper.a [...]

Good job demetrius, so i need to add manually in my platform folder? Any chance you can write a little tutorial to explain which step we need to do? Time to edit your readme :)

demetrius-tech, 2020-07-13:
Thanks @Atomico001! I still hope to find a way to avoid using native code. 

You will find the generic tutorial on how to add native code to NS plug-in here: https://docs.google.com/document/d/1_PDEkC3EtZCsXhthGCmGwtCVVxOwdAepcJaUaA0Uk6k/edit?usp=sharing

Please comment if something is not clear or any issues.

Atomico001, 2020-07-14:
Ty for reply @demetrius-tech . So if i understand right from your guide, i need to do only step 3 and 4. You did the step 1 and 2, right?

Atomico001, 2020-07-14:
Thanks @demetrius-tech unlucky when i try to install your fork i receive this error when try to start nfc reader:
```
00000002 816faa00 -[NFCHardwareManager queueReaderSession:sessionConfig:completionHandler:]:98  error=Error Domain=NFCError Code=202 "Session invalidated unexpectedly" UserInfo={NSLocalizedDescription=Session invalidated unexpectedly}

00000002 838f6300 -[NFCTagReaderSession beginSessionWithConfig:]:324  error:Error Domain=NFCError Code=202 "Session invalidated unexpectedly" UserInfo={NSLocalizedDescription=Session invalidated unexpectedly}, errorCode: 0xca
```

Atomico001, 2020-10-20:
I realy need this, someone can help me?

demetrius-tech, 2020-10-20:
@Atomico001: I would like to help you but I can't find time. As I mentioned before I created a library in ObjectiveC and imported it to the plug-in. But I am not sure it is the best way, so I am looking for a better way to do this, may be a POD or something. I will try to allocate some time to this soon, will let you know.

Atomico001, 2020-10-20:
@demetrius-tech nice to see you again. If you can do a working simple demo with your fork i can investigate it and try myself. Unlucky i receive many error when i tried to import your fork.

Thanks very much!

PaulHieltjesCortegs, 2020-12-28:
> @Atomico001: you can find the ObjC code here: https://github.com/demetrius-tech/nativescript-nfc/tree/dev-ios-write-uid/src/platforms/ios/xcode/NfcNativeHelper
>
> Created the libNfcNativeHelper.a [...]

So I tried the same thing as you and was also stuck on the issue that identifier is always null. Then I read this and thanks to your code got it working. Thank you very much!!!
Your typescript code did generate runtime errors on my device though so I ended up using my own code but with your iOS native code package. 
Again, can't thank you enough for sharing your code here :)