EddyVerbruggen/nativescript-nfc #61: iOS 14 / AppStore : NDEF is disallowed

leganz, 2020-09-25:
## Package.json

```json
"tns-ios": {
      "version": "6.5.2"
}
```

```json
"dependencies": {
    "@angular/animations": "~8.0.0",
    "@angular/common": "~8.0.0",
    "@angular/compiler": "~8.0.0",
    "@angular/core": "~8.0.0",
    "@angular/forms": "~8.0.0",
    "@angular/http": "~8.0.0-beta.10",
    "@angular/platform-browser": "~8.0.0",
    "@angular/platform-browser-dynamic": "~8.0.0",
    "@angular/router": "~8.0.0",
    "@nativescript/webpack": "^3.0.4",
    "@nota/nativescript-webview-ext": "^6.5.4",
    "manifest-android": "^0.1.1",
    "moment": "^2.28.0",
    "nativescript-angular": "^8.21.0",
    "nativescript-appversion": "^1.4.4",
    "nativescript-barcodescanner": "^3.4.2",
    "nativescript-calendar": "^2.2.1",
    "nativescript-dev-version": "^0.2.0",
    "nativescript-forgm-sound": "^1.1.1",
    "nativescript-local-notifications": "^4.2.1",
    "nativescript-localize": "^4.2.2",
    "nativescript-material-activityindicator": "^3.3.0",
    "nativescript-material-button": "^2.5.4",
    "nativescript-material-cardview": "^2.5.4",
    "nativescript-material-core": "^2.5.4",
    "nativescript-material-dialogs": "^2.5.4",
    "nativescript-material-floatingactionbutton": "^2.5.4",
    "nativescript-material-progress": "^2.5.4",
    "nativescript-material-ripple": "^2.5.4",
    "nativescript-material-snackbar": "^2.5.4",
    "nativescript-material-textfield": "^2.5.4",
    "nativescript-nfc": "^4.1.0",
    "nativescript-permissions": "^1.3.9",
    "nativescript-plugin-universal-links": "^1.0.2",
    "nativescript-preferences": "^1.2.0",
    "nativescript-theme-core": "^1.0.6",
    "nativescript-webview-utils": "^3.0.1",
    "plist": "^3.0.1",
    "reflect-metadata": "~0.1.12",
    "rxjs": "^6.6.3",
    "tns-core-modules": "^6.5.18",
    "url": "^0.11.0",
    "zone.js": "~0.9.1"
  },
"devDependencies": {
    "@angular/compiler-cli": "~8.0.0",
    "@ngtools/webpack": "~8.0.0",
    "nativescript-dev-webpack": "^1.5.1",
    "node-sass": "^4.14.1",
    "tns-platform-declarations": "^6.5.15",
    "typescript": "~3.4.0",
    "url": "^0.11.0"
  },
```

I followed your tutorial in the readme. When I try to upload a version of it into the app store (testflight)
I got the following error message. 

````
ERROR ITMS-90778: "Invalid entitlement for core nfc framework. The sdk version '14.0' and min OS version '13.0' are not compatible for the entitlement 'com.apple.developer.nfc.readersession.formats' because 'NDEF is disallowed'."
````

Any ideas what I make wrong?

EddyVerbruggen, 2020-09-25:
Sounds like you're doing it right, but iOS 14 changed something, so something may need to be changed in this plugin as well.

leganz, 2020-09-25:
Thx for the quick respond! I think it might not be a problem in your wonderful package.
I found serveral entries with the same issue.
eg. https://developer.apple.com/forums/thread/123483

closed 2020-09-29 by leganz