EddyVerbruggen/nativescript-nfc #66: NFCNDEFTag.prototype.queryNDEFStatusWithCompletionHandler is undefined

demetrius-tech, 2021-01-08:
Need help please.

I am trying to call the NFCNDEFTag.prototype.queryNDEFStatusWithCompletionHandler method but it is always undefined. Same issue for similar methods which are called from a prototype. 

```
NFCNDEFTag.prototype.queryNDEFStatusWithCompletionHandler.call(tag, (status: NFCNDEFStatus, number: number, error: NSError) => {
     ...
}
```

Error: 

***** Fatal JavaScript exception - application has been terminated. *****
NativeScript encountered a fatal error: Uncaught TypeError: Cannot read property 'call' of undefined
at
push.../../plugins/nativescript-nfc/fork/nativescript-nfc/src/nfc.ts.NFCNDEFReaderSessionDelegateImpl.processNDEFTag(file:///app/0.js:576:67)
at (file:///app/0.js:553:19)
*** Terminating app due to uncaught exception 'NativeScript encountered a fatal error: Uncaught TypeError: Cannot read property 'call' of undefined
at
push.../../plugins/nativescript-nfc/fork/nativescript-nfc/src/nfc.ts.NFCNDEFReaderSessionDelegateImpl.processNDEFTag(file:///app/0.js:576:67)
at (file:///app/0.js:553:19)
', reason: '(null)'
*** First throw call stack:
(0x1b0c759d4 0x1c4626b54 0x1057f4034 0x105bcc0e8 0x105bcbf40 0x105ad0b70 0x1059f6bd4 0x1059f62ac 0x105866cdc 0x1057358ac 0x10582b4e0 0x10582c1b4 0x1b08ad298 0x1b08ae280 0x1b0856dcc 0x1b08578d8 0x1b0861338 0x1f73125a4 0x1f7315874)

cloudhx, 2021-05-10:
Hello @demetrius-tech, I'm running into the same problem and would like to know if you have made any progress on it, thank you very much!

demetrius-tech, 2021-06-25:
> Hello @demetrius-tech, I'm running into the same problem and would like to know if you have made any progress on it, thank you very much!

Somehow it got fixed after upgrading NS to a latest version.

demetrius-tech, 2021-06-25:
Fixed after upgrading NS to a latest version.

closed 2021-06-25 by demetrius-tech

cloudhx, 2021-07-09:
@demetrius-tech Thank you for the update, will try to do the same.

cloudhx, 2021-09-24:
@demetrius-tech I tried both 7 and 8, but cannot make it work, still the same error, my code is below, could you please shed some light on it? Thanks!
```typescript
readerSessionDidDetectTags(
    session: NFCNDEFReaderSession,
    tags: NSArray<NFCNDEFTag> | NFCNDEFTag[]
  ): void {
    const tag = tags[0];
    session.connectToTagCompletionHandler(tag, (error: NSError) => {
      console.log("connectToTagCompletionHandler");

      if (error) {
        console.log(error);
        session.invalidateSessionWithErrorMessage("Error connecting to tag.");
        this.errorCallback(error);
        return;
      }

      const ndefTag: NFCNDEFTag = new interop.Reference<NFCNDEFTag>(
        interop.types.id,
        tag
      ).value;

      try {
        NFCNDEFTag.prototype.queryNDEFStatusWithCompletionHandler.call(
          ndefTag,
          (status: NFCNDEFStatus, number: number, error: NSError) => {
            console.log("queryNDEFStatusWithCompletionHandler");

            if (status == NFCNDEFStatus.NotSupported) {
              var errMessage = "Tag is not NDEF compliant.";
              session.invalidateSessionWithErrorMessage(errMessage);
            } else if (status === NFCNDEFStatus.ReadOnly) {
              var errMessage = "Tag is read only.";
              session.invalidateSessionWithErrorMessage(errMessage);
            } else if (status === NFCNDEFStatus.ReadWrite) {
              const ndefMessage = this._owner.get().message;
              NFCNDEFTag.prototype.writeNDEFCompletionHandler.call(
                ndefTag,
                ndefMessage,
                (error: NSError) => {
                  if (error) {
                    console.log(error);
                    session.invalidateSessionWithErrorMessage("Write failed.");
                    this.errorCallback(error);
                  } else {
                    if (
                      ndefMessage.records[0].typeNameFormat ==
                      NFCTypeNameFormat.Empty
                    ) {
                      session.alertMessage = "Erased data from NFC tag.";
                    } else {
                      if (this.options.writeHint) {
                        session.alertMessage = this.options.writeHint;
                      }
                      this.resultCallback(NfcHelper.ndefToJson(ndefMessage));
                    }
                    session.invalidateSession();
                  }
                }
              );
            }
          }
        );
      } catch (e) {
        console.log(e);
        session.alertMessage = "error";
      }
    });
  }