EddyVerbruggen/nativescript-nfc #70: Cannot read NDEF tag on iOS with ns 8

luffa99, 2021-04-11:
Hi,
I'm testing this essential ns library on both iOS and Android. The following simple app (HelloWorld from ns create)
`import { Observable } from '@nativescript/core'
var Nfc = require("nativescript-nfc").Nfc;
var nfc = new Nfc();
nfc.available().then(function (avail) {
  console.log(avail ? "Yes" : "No");
});
export function createViewModel() {
  const viewModel = new Observable()
  viewModel.message = "-";

  viewModel.onTap = () => {
    viewModel.set('message', '-')
  }

  nfc
  .setOnNdefDiscoveredListener(function (data) {
    console.log(data.message[0]);
    viewModel.set('message', data.message[0].payloadAsString);
  })
  .then(function () {
    console.log("OnNdefDiscovered listener added");
  });

  return viewModel
}
`
does perfectly work on Android, but not on iOS. On iOS it only appears the message "Ready to scan" and nothing happens (I've tested my NDEF tag with other apps and they are able to read it).
After following some hint in https://github.com/EddyVerbruggen/nativescript-nfc/issues/67 there is a bit change: now the OS feedback me after a scan with a small vibration. However the text does not change and nothing is printed in the console.

Thank you

luffa99, 2021-04-18:
I've fixed by using
`Utils.executeOnMainThread`

So I've changed these lines (in _Nfc.prototype.setOnNdefDiscoveredListener_ )
```
//execute on the main thread with this trick, so UI updates are not broken
Promise.resolve().then(function () { return callback(data); });
```
with
```
//execute on the main thread with this trick, so UI updates are not broken
Utils.executeOnMainThread(() => {
    return callback(data); 
})
```
Note that you need first to import Utils from nativescript core.
Note also that you need to comment out entirely `NFCNDEFReaderSessionDelegateImpl.prototype.readerSessionDidDetectTags`
as described in #67

b8ne, 2021-12-07:
@luffa99 Are you able to post exactly how you did this? Was it directly in node_modules or did you pull the files out.
I've applied this under node_modules, but getting Nfc is not a constructor errors now, Im assuming from the Utils import.

BenShelton, 2022-05-24:
It looks like a PR was opened to fix this but was eventually closed: https://github.com/EddyVerbruggen/nativescript-nfc/pull/71

I'm using `patch-package` to fix this locally, see [here](https://github.com/ds300/patch-package#readme) for installation instructions, and then you can add the following patch file in `patches/nativescript-nfc+5.0.0.patch`:

```diff
diff --git a/node_modules/nativescript-nfc/nfc.ios.js b/node_modules/nativescript-nfc/nfc.ios.js
index c55b377..c300242 100644
--- a/node_modules/nativescript-nfc/nfc.ios.js
+++ b/node_modules/nativescript-nfc/nfc.ios.js
@@ -1,6 +1,7 @@
 "use strict";
 Object.defineProperty(exports, "__esModule", { value: true });
 exports.Nfc = void 0;
+var core_1 = require("@nativescript/core");
 var nfc_common_1 = require("./nfc.common");
 var Nfc = /** @class */ (function () {
     function Nfc() {
@@ -53,8 +54,9 @@ var Nfc = /** @class */ (function () {
                             JSON.stringify(data));
                     }
                     else {
+                        core_1.Utils.executeOnMainThread(() => callback(data))
                         // execute on the main thread with this trick, so UI updates are not broken
-                        Promise.resolve().then(function () { return callback(data); });
+                        // Promise.resolve().then(function () { return callback(data); });
                     }
                 }, options);
                 _this.session = NFCNDEFReaderSession.alloc().initWithDelegateQueueInvalidateAfterFirstRead(_this.delegate, null, options && options.stopAfterFirstRead);
@@ -125,7 +127,7 @@ var NFCNDEFReaderSessionDelegateImpl = /** @class */ (function (_super) {
         // execute on the main thread with this trick
         this.resultCallback(NFCNDEFReaderSessionDelegateImpl.ndefToJson(firstMessage));
     };
-    NFCNDEFReaderSessionDelegateImpl.prototype.readerSessionDidDetectTags = function (session, tags) {
+    // NFCNDEFReaderSessionDelegateImpl.prototype.readerSessionDidDetectTags = function (session, tags) {
         /*
         // TODO prolly remember the tags for when the app wants to write to it (also: check Android impl for possibly sth similar)
         const nfcNdefTag = tags[0];
@@ -146,7 +148,7 @@ var NFCNDEFReaderSessionDelegateImpl = /** @class */ (function (_super) {
           });
         }
        */
-    };
+    // };
     // Called when the reader session becomes invalid due to the specified error
     NFCNDEFReaderSessionDelegateImpl.prototype.readerSessionDidInvalidateWithError = function (session /* NFCNDEFReaderSession */, error) {
         this._owner.get().invalidateSession();
```