EddyVerbruggen/nativescript-nfc #72: Callback not initiating on ios 14.6 with ns 8

AchrafBardan, 2021-06-21:
This is my code 
```typescript
new  Nfc().setOnNdefDiscoveredListener(
      (data: NfcNdefData) => {
        console.log(data)
      },
      {
        stopAfterFirstRead: true
      }
    ).then(data => {
      return data;
    })
    .catch(() => {
      throw new Error("nfc fail")
    }
    );
```

When i run this the visual nfc scan view gets showen
[image]

But nfc cards are not being scanned this is showen in the console:
```
00000002 83138630 -[NFCNDEFReaderSession beginSession]:293[NFCNDEFReaderSessiondidUIControllerInvalidate:]:187  sessionState=2, proxy=YES
```

Dankje Eddy voor alle contributies die je hebt gedaan voor de NativeScript community

b8ne, 2021-12-07:
Has anyone got a solution for this yet?

mastashake08, 2021-12-09:
Nope and I need one
*Jyrone Parker*
*Technical Consultant, *Technopreneur

*Never Let Limited Minds Limit Your Success*
*https://www.jyroneparker.com <http://www.jyroneparker.com> |
https://youtube.com/c/jyroneparker <https://youtube.com/c/jyroneparker>*
Twitter: @mastashake08

BenShelton, 2022-05-24:
This looks to be a duplicate of #70, solutions are there